logisim-evolution/logisim-evolution #1524: RGB video clear pixel dot

RomaricKc1, 2022-08-23:
I may not understand it but it seems impossible to clear a pixel dot written on the RGB video. 
I use monochrome 4-bit as color on ```128x64```. 

When I apply logical ```1``` on the ```reset```, it darkens the entire display. And I did not find a way to clear a dot after it has been written into the display (using WE pin).
I thought the reset pin should only remove the pixel written on a specific ```(x, y)``` location. Not darken the screen (it goes from yellow to black).

As of now, the only way (I've found) to remove an unwanted pixel dot is to hit ```Ctrl+rR``` to reset the circuit. Maybe there will be another IO for clearing an already set pixel dot?
PS:  I use asynchronous and synchronous, both darken the display and do not remove an already set pixel dot.

BFH-ktt1, 2022-08-24:
RGB Video is a component that I merged in the last years from @kevinawalsh from his holycross version of logisim. No testing has been done and it might well be that there are some bugs due to the evolution of the holycross fork and the evolution changes. For the moment I cannot provide support for this component (same as the components in the "Input/Output Extra" library that were merged from the ITA fork). They are there "without any guarantee", sorry about that. I dig into them when my time permits.

RomaricKc1, 2022-08-24:
Alright, thank you for taking a look at it.
I'll keep an eye on it.

RomaricKc1, 2022-08-26:
Anyone interested, I have this project that uses the RGB component. [See](https://github.com/RomaricKc1/rgb_video_card).

closed 2022-08-26 by RomaricKc1