logisim-evolution/logisim-evolution #1531: Saving RAM image file performance problem (maybe caused by preview pane)

kjcolley7, 2022-08-28:
When right clicking a RAM component and using the "Save image..." option, the dialog is super laggy and my laptop's fans come on. I'm pretty sure this is caused by the RAM being 32MB, and the fact that the dialog shows a "preview" of the data (which is actually the entire contents of RAM shoved into a text area). When I changed the output format to binary data (big endian), it took a few minutes for the UI to unfreeze, at which point the preview section of the save dialog was "helpfully" showing the binary data (mostly as Unicode missing characters).

The preview section of the save image dialog should probably only show a preview of the first ~1KB of the image, not the entire thing.

Using Logisim Evolution 3.7.2 (from brew) on macOS 12.3.1.

BFH-ktt1, 2022-08-28:
Thank you for your issue. Logisim was not intended for such big RAM components, and hence the underlying java structures slow it down drastically. I think that even having this "reduced vision" that it will be tedious slow also. Optimizing logisim for speed is out of the scope of our roadmap, hence PR's are welcome, but it is unlikely that we will work on this.

dtowell, 2022-09-13:
@kjcolley7 I'll investigate. No promises.