logisim-evolution/logisim-evolution #1538: Can't import my jar library

MicalProger, 2022-08-31:
Hello. Firstly, in logisim logisim-evolution in tutorial, code of example is for logisim and isn't usefull for logisim-evolution. But main path of this issue is what build of my library is not importing to my project. After selection of my jar new tab with my components is not creating. That menu stays the same. Maybe you can give me some code as example of working library. For this version of my lib i used code in com.cburch.gray.

R3dst0ne, 2022-09-01:
`com.cburch.gray` is very outdated.
Take a look at the [std libraries](https://github.com/logisim-evolution/logisim-evolution/tree/develop/src/main/java/com/cburch/logisim/std). They are up-to-date and work the same way as external ones.

MicalProger, 2022-09-01:
Ok. I  whatched this new examples. I saw what in new version AddTool  are not used, but user just Tool ,tools list makes from FactoryDescription[] DESCRIPTIONS, for string i should use Strings.S.getter. But it still don't works. Maybe i should do something more?

R3dst0ne, 2023-02-05:
Close because of inactivity?

closed 2023-03-03 by VoxelTube