logisim-evolution/logisim-evolution #1544: Clock Frequency, not reaching 2048K

Alexandros444, 2022-09-23:
Hello Logisim-Community, 

I am currently designing an 8-Bit-PC, and for the IC-Simulation I used Logisim-Evolution.

I set the Frequency in Logisim to be 2048KHz, but only got around 40Hz (in the top left corner). 
Why is that?

Is it too demanding for the Hardware? (My most stressed CPU-Core sits at around 60%)

I quickly slammed some gates together to illustrate my point:
[image]
Here the Clock is at around 400Hz, even tough it is set to 2048KHz and the circuit is fairly simple.

Thanks for an answer

Best regards
Alexander Mohr

alexandre-lecoq, 2022-09-24:
It's the max speed at which your machine can simulate this circuit.

Logisim could probably be optimized to run simulations faster, but it has still to be done.

maehne, 2022-09-25:
Closing as duplicate of issues #582 and #786.

closed 2022-09-25 by maehne