logisim-evolution/logisim-evolution #1546: Command-line verification cleanup

dtowell, 2022-09-25:
The existing command-line verification features appears to be somehow haphazard. The documentation is at least incomplete and possibly misleading. Comments in the code suggest the existing command-line options are not organized well. Some options affect other options in ways only visible by reading the code. 

I am interested in rewriting (refactoring) the existing verification functionality (as I use it regularly). My priorities would be to improve (English) documentation, untangle the command-line options, and generally make the features useful for non-GUI verification tasks (like homework checking). In addition the code seems to suffer from feature creep and generally needs some clean up. A good solution would probably necessitate some command-line-options breaking changes. 

Is this worth pursuing? Are there non-negotiable constraints I might not be aware of?

R3dst0ne, 2022-09-25:
A year ago, we partly reworked the command line options.
Have a look at the issues #963, #981 and the PR #983.

The discussion #598 is about improving the test vector component to add better support for sequential circuits testing.

The bad state of the documentation is a problem in general. So every help with it is more than welcome.

Before starting this endeavor, it might be a good to discuss this topic and collect ideas from people that use it / are interested in using it.

maehne, 2022-09-25:
@dtowell: Help with improving the documentation and the verification functionality is very welcome! I agree with @R3dst0ne that it would be good if you would detail here your planned changes in more detail so that we can provide feedback before you start with the implementation.

dtowell, 2022-09-26:
@maehne et al, I will be happy to do that. I don't mean to assume blanket coverage for any change. Once I investigate a bit further I will post a detailed plan (or at least goal).

R3dst0ne, 2022-10-12:
Found an additional issue about the command line options.
#986

maehne, 2022-10-12:
@dtowell: Thanks for your analysis and willingness to clean up these problems! I am looking forward to your PR(s)!

dtowell, 2022-10-15:
@maehne @R3dst0ne please look over #1563 and provide feedback. I realize this is a large change. Does this look promising? Or, should I backup and try again?

maehne, 2022-10-27:
@dtowell: PR #1563 goes for me in the right direction.

dtowell, 2022-11-05:
#1581 should be good to review and hopefully integrate.

maehne, 2026-08-19:
@hewzhew: Could you please have a look at closed PR #1546 by @dtowell and see what can be integrated into the current code base? Unfortunately, it got stuck at some point in the review process even though only minor things were left to be resolved. I think the work done there is still valuable and hope that it has not too many conflicts with the many recent changes.

hewzhew, 2026-08-19:
Thanks for the pointer. I believe the intended closed PR is #1581, which superseded the Draft #1563. I compared its final head with current main. The whole branch is now 2491 commits behind and its 24-file patch overlaps heavily with today's `Main`, `Startup`, `TtyInterface`, `Download`, and locale resources, so rebasing or cherry-picking it wholesale would be unsafe.

The parse-time status part is already covered by merged #2836 and #2850. The most reusable remaining ideas are making `TtyInterface` return a status instead of calling `System.exit()` internally, using symbolic statuses, separating startup modes, and restoring runtime test coverage. However, #1581 also changes observable behavior: it uses argument status 1 instead of the current 10 and rejects several combinations that current main accepts, including GUI substitutions, multiple TTY files, GUI-only options with TTY, and mixed TTY/test modes.

I propose first taking a behavior-preserving slice that only centralizes TTY exit ownership and adds focused tests while retaining current numeric statuses and accepted combinations. A second slice could separate startup modes, still without policy changes. We could then discuss each option-combination restriction separately, and keep the command-line documentation update independent. Ready PR #2854 overlaps `Startup`/download handling, so I would account for its exact cable-selection contract before implementing the first slice. Does this sequence match what you had in mind?

maehne, 2026-08-21:
@hewzhew: Your suggested approach makes sense to me.