logisim-evolution/logisim-evolution #1548: VHDL Simulation does not work on Windows, LM_LICENSE_FILE

marsfan, 2022-09-27:
This is similar to #1442, but is on Windows so I figured it might be best to open a separate issue. 

I installed Quartus and obtained a free license for Questa. I set the `LM_LICENSE_FILE` environment variable in Windows to point to the license file, and confirmed that I can launch Questa (both by clicking on it in the start menu, and by running `vsim` in a command line). 

<details>
<summary> Environment Variable Screenshot</summary>

[image]

</details>

I also confirmed that I can access the environment variable from the terminal in both PowerShell and Command Prompt:

PowerShell: 
```PowerShell
PS C:\Users\Gabe Roper> $env:LM_LICENSE_FILE
C:\intelFPGA_lite\questa_license.dat
```

Command Prompt:
```cmd
C:\Users\Gabe Roper> echo %LM_LICENSE_FILE%
C:\intelFPGA_lite\questa_license.dat
```

However, when I set up a VHDL entity in Logisim and enable VHDL simulation, I get the following message both as a pop-up and in the VHDL simulation log: 

```
Unable to checkout a license. Make sure your license file environment variable (e.g., LM_LICENSE_FILE)
is set correctly and then run 'lmutil lmdiag' to diagnose the problem.
Unable to checkout a license.  Vsim is closing.
** Error: Invalid license environment. Application closing.
```

I have tried setting the environment variable both in both the "User Environment Variables" section and the "System Environment Variables" section, but I get the same result either way. 

## System Information
* Windows 11 21H2
* Logisim-Evolution 3.7.2 (installed using MSI installer)
* Questa - Intel FPGA Starter Edition 21.1.1.850 (Free License version that can be obtained from Intel)

marsfan, 2022-09-27:
Despite not knowing Java, I figured that I would poke around at the source code, and I think I may have isolated part of the problem, though I could be wrong. 
In the file [./src/main/java/com/cburch/logisim/vhdl/sim/VhdlSimulatorTclBinder.java](./src/main/java/com/cburch/logisim/vhdl/sim/VhdlSimulatorTclBinder.java), on line 72, the is the following:

```java
    env.put("LM_LICENSE_FILE", "1650@eilic01");
```

So it looks like the value for the license file has been hard-coded into the source, instead of having it use the environment variable that I set in my system.

maehne, 2022-09-27:
@marsfan: Thanks a lot for your detailed bug report and for finding the source of the problem! Indeed, `LM_LICENSE_FILE` should not be hard-coded in the sources. We never noticed it, as we only tested it using ModelSim Altera Starter Edition, which didn't need a license. Now that Quartus comes with QuestaSim Starter Edition, which always requires a license, this bug becomes more urgent.

BFH-ktt1, 2022-10-02:
Closing as resolved by my latest PR. Please reopen if not resolved.

closed 2022-10-02 by BFH-ktt1

marsfan, 2022-10-06:
@BFH-ktt1 @maehne I just updated to Logisim 3.8.0 and the VHDL simulator seems to be working now. 

There are two small details that I noticed that might be worth documenting somewhere:
1. If "VHDL Simulation Enabled" is not checked in the "Simulate" Menu, then placing VHDL entities will cause the text "Simulation halted by internal error" to appear on the schematic. 
2. The VHDL simulator does not work if the project has never been saved.

marsfan, 2022-10-06:
From further testing, it appears I was mistaken about 2. It does work if the project was not saved. 
It seems that the component I was testing uses some syntax that Questa does not like.

maehne, 2022-10-06:
Thanks for the feedback @marsfan! Could you please file a new issue about 1 so that we don't loose track of it? The error message in that case deserves to be improved.

marsfan, 2022-10-11:
Sure. Going to play around with it for a few days first to see if I can find any other stuff to file.