logisim-evolution/logisim-evolution #1569: Debian build uses zstd compression which is incompatible with dpkg

tiotp, 2022-10-23:
Installing the provided .deb file in a debian 11 virtual machine returns 

`dpkg-deb: error: archive '/home/user/Downloads/logisim-evolution_3.8.0-1_amd64.deb' uses unknown compression for member 'control.tar.zst', giving up`

This is a documented issue for debian, and can likely be resolved by changing the compression method of control.tar.

Thank you!
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892664

maehne, 2022-10-26:
Thanks for reporting this issue. I believe this is due to the fact that the DEBs are built on Ubuntu. 

@MarcinOrlowski: Could we maybe switch the build flows from Ubuntu to Debian for better compatibility?

MarcinOrlowski, 2022-10-26:
@maehne I think we can.  Will look into it in the evening, esp. I planned also to see why *.snap builder is struggling.

maehne, 2022-12-15:
Issue was again hit in discussion #1600.

MarcinOrlowski, 2022-12-17:
Sorry, haven't had much spare cycles recently. Hope to look at it over the xmas break, esp. I also need to have to take care of the Snaps.

maehne, 2024-10-23:
Also issue #2061 shows the interest in a DEB package working on Debian.

maehne, 2026-08-19:
@anrouxel: Could you maybe also have a look at this issue, as you are anyway currently touching our GitHub actions?

anrouxel, 2026-08-19:
@maehne I have a question. What is the minimum version of Debian and Ubuntu that you plan to support via the deb format?

maehne, 2026-08-21:
@anrouxel: Thanks for looking into this.

IMHO, we should at least support the current [Debian stable release](https://www.debian.org/releases/), i.e., Trixie. Oldstable Bookworm is end of life since July. Whether we would additionally like to target testing is a question of effort.

MarcinOrlowski, 2026-09-05:
As `dpkg` supports zstd since v1.21, shall we close this ticket as it seems no longer relevant (also +1 to not support legacy)

MarcinOrlowski, 2026-09-05:
@anrouxel Current *.deb shall be installable on Ubuntu 24.04+ and Debian 13 (Trixie)

MarcinOrlowski, 2026-09-05:
I created separate ticket to for the `t64` related issue so we are good here

@SamBkamp please do not hijack existing tickets. Creating new one is really simple and quick and by doing so you are not adding unrelated noise to existing discussion. Each problem shall have separate ticket for sake of handling.

MarcinOrlowski, 2026-09-05:
After #2959 is merged, Ubuntu 22.04 and Debian 12 will also be able to install form our `*.deb`

maehne, 2026-09-05:
Closing this as resolved, as suggested by @MarcinOrlowski.

closed 2026-09-05 by maehne