logisim-evolution/logisim-evolution #1572: Ram outputs X on first clock cycle

cheyao, 2022-10-31:
When i reset the ram, it outputs X before the first clock, how can i disable this? its very annoying

[image]

davidhutchens, 2022-11-02:
The RAM defaults to synchronous read (i.e. Asynchronous read = No). Thus no output occurs until a rising edge. If you switch the Asynchronous read attribute to Yes, it will output a value before the rising edge, and switch outputs as soon as the control or address lines switch.

Is that what you were looking for?

cheyao, 2022-11-02:
> The RAM defaults to synchronous read (i.e. Asynchronous read = No). Thus no output occurs until a rising edge. If you switch the Asynchronous read attribute to Yes, it will output a value before the [...]

Ohh yes that was what i was looking for! thanks a lot! 

And do you know what byte enables and line enables mean? thx

davidhutchens, 2022-11-02:
To see what the Enables attribute does, please follow this tour.

Start a new project and place a single RAM module on it. It should have Enables: Use byte enables.

Change the Data Bit Width to 32.

Note there is an attribute Read write control. Switch that from Whole word read/write to Use byte enables. Note that you now have write control lines for each of the 4 bytes of the 32-bit word. This allows byte addressable 32-bit words.

Switch Read write control back to Whole word read/write.

Switch Enables to Use line enables. Note that the Read write control attribute disappeared and a Line size attribute is now available. Switch Line size to Octo. There are now 8 Line enable control lines. Also note that there are now 8 input and 8 output data lines, each 32 bits. You can actually read or write up to 8 32 bit words at one time. This models things like a cache line of 8 words.

cheyao, 2022-11-02:
Ohhhhh thanks a lot for the explination!

davidhutchens, 2022-11-02:
We are happy to help. If all questions are resolved, please close this issue as complete.

cheyao, 2022-11-02:
Yes everything is resolved, thanks again! (And sorry i forgot to close the issue 😅)

closed 2022-11-02 by cheyao