logisim-evolution/logisim-evolution #1577: Using a Shift-registers output invalid Verilog

89Mods, 2022-11-04:
I have the following schematic:
[image]

When I try to simulate the Verilog generated by Logisim for this module, iverilog throws the following error on build:
`/home/tholin/tt2-avalonsemi-5401/src/circuit/clock_gen.v:121: error: Net open is not defined in this context.
/home/tholin/tt2-avalonsemi-5401/src/circuit/clock_gen.v:121: error: Output port expression must support continuous assignment.
/home/tholin/tt2-avalonsemi-5401/src/circuit/clock_gen.v:121:      : Port 4 (q) of Shift_Register is connected to {open, s_logisimNet7, s_logisimNet8, s_logisimNet2}`

Looking at the generated Verilog, the issue appears to be here, where the shift register module is used:
`
   Shift_Register #(.negateClock(0),
                    .nrOfBits(1),
                    .nrOfParBits(4),
                    .nrOfStages(4))
      MEMORY_3 (.clock(s_logisimNet3),
                .d({s_logisimNet15,
                    s_logisimNet14,
                    s_logisimNet13,
                    s_logisimNet12}),
                .parLoad(s_logisimNet0),
                .q({open,
                    s_logisimNet7,
                    s_logisimNet8,
                    s_logisimNet2}),
                .reset(s_logisimNet11),
                .shiftEnable(s_logisimNet9),
                .shiftIn(s_logisimNet1),
                .shiftOut(s_logisimNet1),
                .tick(1'b1));
`

In particular, the output list:
`
                .q({open,
                    s_logisimNet7,
                    s_logisimNet8,
                    s_logisimNet2}),
`
There is no net called 'open'. Infact, this entry is superfluous, and does not map to any part of my schematic. It can safely be removed.

Additionally, not connecting one of the outputs of the shift register causes the Verilog generator to insert a line with nothing but a single comma in it, also causing a compiler error:
`
                .q({open,
                    s_logisimNet7,
                    ,
                    s_logisimNet2}),
`
This forced me to use a "no connect" symbol.

maehne, 2022-11-04:
Thanks for reporting the bug. Verilog netlisting is unfortunately less tested than VHDL netlisting. We'll look into it, but it may take some time.

BFH-ktt1, 2022-11-15:
@89Mods, could you please attach the circuit (as a .zip file) such that I can debug the problem?

89Mods, 2022-11-15:
@BFH-ktt1 
[i1577.zip](https://github.com/logisim-evolution/logisim-evolution/files/10013220/i1577.zip)

Here you go. This will replicate both issues I mentioned if exported to Verilog.

BFH-ktt1, 2022-11-15:
@89Mods : The problem is a bit more complicated and related to following point:
In VHDL you can assign each bit of an output port separately, and use the keyword "open" if it is not connected.
In Verilog, you cannot assign each bit (to my knowledge), and hence have to do a concatenation the {open,,,s_logisimNet2}.
This works for Intel Quartus.
Changing it in {open,open,open,s_logisimNet2} works than in Modelsim, it will just "introduce" the new wire "open", that of course creates a short-circuit on those bits....
Iverilog is even stricter and throws an error as the wire open is not defined.
The clean solution to this problem would be to introduce for Verilog a dummy bus in which some bits are not connected to anything else. This problem is most probably not only valid for the shift-register, but also to other modules. 
It will take me a bit of time to "fix" this bug.
For the moment you can "fix" it by using your solution with the wire wit a "don't connect" and add in the circuit where the module is instantiated:
wire open;
Thanks for this bug report.