logisim-evolution/logisim-evolution #1587: it makes a error when i try to run it. is therer a way to fix it?

utubysdog, 2022-11-11:
i run "java -jar logisim-evolution-3.8.0-all.jar" an it spits out this error.

`Error: LinkageError occurred while loading main class com.cburch.logisim.Main
        java.lang.UnsupportedClassVersionError: com/cburch/logisim/Main has been compiled by a more recent version of the Java Runtime (class file version 60.0), this version of the Java Runtime only recognizes class file versions up to 55.0`

maehne, 2022-11-11:
The error message tells you that your JDK/JRE version is older than the one used to build Logisim-evolution. You should use at least OpenJDK 16. However, I would recommend you to use one of the provided installers that matches your operating system. They package Logisim-evolution with all dependencies including a stripped down JRE. They also provide integration with your graphical desktop environment.

maehne, 2022-11-15:
Closing this issue, I guess it gets resolved by following my above comment. Feel free to reopen if you still have problems.

closed 2022-11-15 by maehne