logisim-evolution/logisim-evolution #1589: Windows package not included in Nightly

davidhutchens, 2022-11-25:
The Windows .msi package was not included in the nightly after #1588 was merged.

Perhaps I didn't guess the osArch string correctly and the package name isn't what I expected.

@BFH-ktt1: Would you please post the precise package file names that are generated by the current develop branch for the four combinations (macOS, Windows) x (Intel, Arm) so I can fix the nightly and the README?

BFH-ktt1, 2022-12-01:
@davidhutchens : I hope to take some time this weekend to look into it, sorry, I have been quite busy lately :-)

maehne, 2022-12-15:
@BFH-ktt1: Ping

davidhutchens, 2023-01-12:
@BFH-ktt1: Could you please provide the filenames I requested previously so this issue can be resolved?

BFH-ktt1, 2023-01-18:
@davidhutchens : Windows intel: logisim-evolution-3.9.0-amd64.msi

davidhutchens, 2023-01-18:
@BFH-ktt1: Thank you!

davidhutchens, 2023-01-18:
Closing as completed with #1645.

PR #1645 (2023-01-18): Fix package names for Mac and Windows packages
Fixes #1589.
closed 2023-01-18 by davidhutchens