logisim-evolution/logisim-evolution #1593: "A DLL required for this install to complete could not be run."

sykeben, 2022-11-28:
I'm having issues with the `.msi` Windows package installer for amd64 in which the following error message appears after clicking "next" on the "Destination Folder" step in the installer:
[image]

Is this an issue with the installer itself, or should I run a system repair tool like `sfc /scannow`?

davidhutchens, 2022-11-28:
To help us track down the problem, could you please tell us where you got the .msi file?

Also, please post its full name here. That should include version information.

What version of Windows are you running?

sykeben, 2022-11-28:
I feel silly. I misread the end of the filename on the releases page as `amd64` when in fact it was `arm64`. I have an x86-based machine, so of course it won't run (wrong architecture). Apologies!

davidhutchens, 2022-11-28:
I assume you grabbed the x86 version and that works for you. It might not be entirely your fault as we should have labeled the x86 version with x86_64 or amd64. It should be the 64 bit version.

If that works for you, please close this issue as complete.