logisim-evolution/logisim-evolution #1594: Wire outputting a value of E instead of 1

Alice-Cheshire, 2022-11-29:
This might be a matter of user error since I'm fairly new to learning logic gates and using Logisim but I'm having trouble finding anything that might explain what's happening here and why. The red wire in this adder/subtracter should only be 0 or 1, not E, since this is a 1-bit device.

Visual: https://i.imgur.com/22LmG88.png

In case they will help, these are the other two circuits I'm using here.
Eight of those adder/subtracters combined into an 8-bit one: https://i.imgur.com/KQpdqF7.png
And my main circuit, along with the current state that's causing this issue: https://i.imgur.com/BJDMd2H.png

davidhutchens, 2022-11-29:
You have tied your SubtractFlag line with your CarryIn line. SubtractFlag is a 1, CarryIn is a 0. That line cannot be both 1 and 0 at the same time. Therefore the simulation marks it as an error. Generally, you should only combine inputs through logic gates.

Alice-Cheshire, 2022-11-29:
Ah, so I should be using an OR gate there then. I'd been following directions I found online which had the two directly connected like that. That does look to have fixed the issue though. I'll have to be more careful about that in the future. Thanks for the help.

As a bit of an aside, if possible it might be better for the program to actually state "Error" instead of just "E" when something like this happens. I'd interpreted it as 0xE in hex instead of "E" for "Error".

maehne, 2022-12-15:
@Alice-Cheshire: Please check `Help -> User's Guide -> Guide to Being a Logisim User -> Additional features -> Wire colors` for an explanation of the different wire colours. We could consider adopting for the signal values Verilog/VHDL conventions (where feasible). @davidhutchens and @BFH-ktt1: What do you think?

davidhutchens, 2022-12-16:
@maehne and @BFH-ktt1: A case can be made to use established conventions when possible. But Logisim and Logisim-evolution have used these signal values for a long time. I am not aware of any significant misunderstanding in our user base. However, those who are teaching students with both LE and VHDL might find it easier if there were consistency here. I have no experience with that.

I have not examined the code to see how hard it would be to change these values. Documentation would also need to be updated. It seems to me the upside might be nice but small and the effort would be relatively large. Therefore, my advice would be to use our time and effort toward other goals. I would rate this as a low priority.

maehne, 2022-12-20:
Thanks for your feedback @davidhutchens. Personally, I would like to see an adjustment to Verilog/VHDL conventions for consistency reasons even though I agree that there's also Logisim's history to be considered.

@BFH-ktt1: What do you think?