logisim-evolution/logisim-evolution #1597: Erro ao nomear um componente

blc-oak, 2022-12-03:
Olá amigos, estpu tendo problemas com o Logisim Evolution, que ao criar uma Tabela Verdade para gerar circuitos, e dar nomes à variáveis, ao ser gerado o novo circuito, algumas entradas e saídas aparecem sem o seu Rótulo. Tento renomear manualmente, mas aparece a mensagem "Já exite um componente com essa etiqueta! Por favor, use etiquetas exclusivas em um circuito." como se ja tivesse um componente com essa etiqueta, mas aparece o mesmo erro, se eu criar um novo projeto e tento criar novamente a tabela verdade. Alguém com mais experiência no Software, sabe como resolver esse problema? Att. Bruno.
[image]

maehne, 2022-12-15:
Please, report any issues in English to ensure that all contributors can understand your problem (and increase the chance for a quicker response)! [DeepL](https://www.deepl.com/) translates your issue to:

> Hello friends, I am having problems with Logisim Evolution, that when creating a Truth Table to generate circuits, and give names to variables, when the new circuit is generated, some inputs and outputs appear without their Label. I try to rename them manually, but the message "A component with this label already exists! Please use unique labels in a circuit." as if there was already a component with that label, but the same error appears if I create a new project and try to create the truth table again. Does anyone with more experience in the Software know how to solve this problem? Att. Bruno.

This description is a bit vague. Please, always describe as precise as possible the steps to reproduce your issue. Could it be that you try to use the same signal name more than once (e.g., as input and output) in your truth table? This is not supported. Signal names need to be always *unique*. Use, e.g., suffixes to distinguish input and output signals, e.g., `state` and `state_next` for a next state logic combinational circuit.

Please give feedback if this resolves your issue.

BFH-ktt1, 2025-06-01:
Closing issue as longtime no action