logisim-evolution/logisim-evolution #1598: Inconsistent behavior of RAM between generated VHDL and logisim

dxhisboy, 2022-12-08:
I'm trying to make use of logisim-evolution for education purpose, when I tried to verify the FPGA workflow I found the behavior of memory in simulation and generated VHDL is different.

For example, a very simple memory test circuit:
[image]
It produces a wave like this in logisim (the data corresponding to addr is read out when clock rises):
[image]
But when I tried to simulate generated VHDL, it seems there is additional delay to response the read request:
[image]
I do know in practice memory should higher latency, but the inconsistency in logisim-evolution and generated VHDL would cause troubles in experiments.

I'm not sure whether the delay is intended or accidental.
So I have the following questions:
1. Do you want to make the behavior consistent?
2. If yes to 1, is there any plan to fix this issue?
3. If no to 2, will a pull request for this issue welcomed?
4. If no to 1, if a pull request to add additional option to generate consistent VHDL welcomed?

dxhisboy, 2022-12-09:
Also, a minor problem is:
For a RAM with 256 entries, it generates:
```
TYPE byteArray IS ARRAY ( 256 DOWNTO 0 ) OF std_logic_vector( 7 DOWNTO 0);
```
But `256 DOWNTO 0` indicates 257 entries.

Should be fixed in `Hdltypes.java:96`, substract `myNrOfEntries` by 1.

BFH-ktt1, 2022-12-09:
@dxhisboy : Concerning your first remark:
This is wanted behavior, as the RAM components normally should be operated with a clock component, and not, as you do, with an input, this is basically seen as a "gated clock", and hence will not, by default, result in a correct behavior. There are some more components that are implemented "multi-cycle".
Concerning your second remark, yes this is a bug, thanks for reporting it.

dxhisboy, 2022-12-10:
> @dxhisboy : Concerning your first remark: This is wanted behavior, as the RAM components normally should be operated with a clock component, and not, as you do, with an input, this is basically seen [...]
I also tested with the clock component, it does not affect the behavior of generated VHDL.

BFH-ktt1, 2022-12-10:
The reasoning here is following:
1) in logisim a logisim clock cycle takes at least 4 real clock cycles.
2) FPGA's have only synchronous RAM components.
3) To be able to model asynchronous RAM component, we omit the extra line of flioflops.
4) To be able to model synchronous ram components we add the line of extra flipflops.
Hence the behavior of the generated HDL is completely in line with the wanted behavior, as the address and data on the inputs will only change each 4 clock cycles.
Connecting directly a clock and change the address and data each clock cycle will, therefore, generate your observations, however, in the generated circuit by logisim using the clock component this will result in a correct behavior of the complete system.
It is therefore no bug, but a wanted behavior.

dxhisboy, 2022-12-10:
> The reasoning here is following:
>
>     1. in logisim a logisim clock cycle takes at least 4 real clock cycles.
>
>     2. FPGA's have only synchronous RAM components.
>
>     3. To be able to model [...]

I don't think there is a chance to simulate asynchronous RAM with synchronous RAM. Imagine this condition:
```
x=>[addr RAM A data]=>[addr RAM B data]=>[addr RAM C data]=>y
```
When simulated asynchronous RAM is connected in a chain, the required cycles is unpredictable.
Asynchronous RAM should be implemented like an array of registers (while this is a waste of hardware resources, but mostly logisim does not work with extreme conditions).

dxhisboy, 2022-12-11:
Also, it seems asynchoronous RAM is not currently supported by logisim, in `RamHdlGeneratorFactory.java`:
```
  @Override
  public boolean isHdlSupportedTarget(AttributeSet attrs) {
    if (attrs == null) return false;
    Object busVal = attrs.getValue(RamAttributes.ATTR_DBUS);
    final var separate = busVal != null && busVal.equals(RamAttributes.BUS_SEP);
    Object trigger = attrs.getValue(StdAttr.TRIGGER);
    final var asynch = trigger == null || trigger.equals(StdAttr.TRIG_HIGH) || trigger.equals(StdAttr.TRIG_LOW);
    final var byteEnabled = RamAppearance.getNrLEPorts(attrs) == 0;
    final var syncRead = !attrs.containsAttribute(Mem.ASYNC_READ) || !attrs.getValue(Mem.ASYNC_READ);
    final var clearPin = attrs.getValue(RamAttributes.CLEAR_PIN) == null ? false : attrs.getValue(RamAttributes.CLEAR_PIN);
    final var readAfterWrite = !attrs.containsAttribute(Mem.READ_ATTR) || attrs.getValue(Mem.READ_ATTR).equals(Mem.READAFTERWRITE);
    return Hdl.isVhdl() && separate && !asynch && byteEnabled && syncRead && !clearPin && readAfterWrite;
  }
```
Also, I tested for the HDL generation in #1601, by moving the data readout procedure out of the clock rising edge controlled if branch, Quartus will try to use registers to implement the asynchronous RAM.

dxhisboy, 2022-12-11:
I tested to move the dataOut path out of clock controlled region in RAM:
```
ARCHITECTURE platformIndependent OF RAMCONTENTS_RAM_1 IS 

--------------------------------------------------------------------------------
-- Here all private types are defined                                         --
--------------------------------------------------------------------------------
   TYPE byteArray IS ARRAY ( 255 DOWNTO 0 ) OF std_logic_vector( 7 DOWNTO 0);

--------------------------------------------------------------------------------
-- All used signals are defined here                                          --
--------------------------------------------------------------------------------
   SIGNAL s_byteMem0Contents : byteArray;
   SIGNAL s_byteMem1Contents : byteArray;
	SIGNAL s_dataout : std_logic_vector(15 downto 0);
BEGIN

   --------------------------------------------------------------------------------
   -- The control signals are defined here                                       --
   --------------------------------------------------------------------------------
   --------------------------------------------------------------------------------
   -- The actual memorie(s) is(are) defined here                                 --
   --------------------------------------------------------------------------------
	dataOut <= s_dataout;
	s_dataout(7 DOWNTO 0) <= s_byteMem0Contents(to_integer(unsigned(address))) when oe = '1' else s_dataout(7 downto 0);
	s_dataout(15 DOWNTO 8) <= s_byteMem1Contents(to_integer(unsigned(address))) when oe = '1' else s_dataout(15 downto 8);
   mem0w : PROCESS(clock, we, byteEnable0, dataIn, address) IS
   BEGIN
      IF (rising_edge(clock)) THEN
         IF (byteEnable0 = '1') THEN
            IF (we = '1') THEN
               s_byteMem0Contents(to_integer(unsigned(address))) <= dataIn(7 DOWNTO 0);
            END IF;
         END IF;
      END IF;
   END PROCESS mem0w;

   mem1w : PROCESS(clock, s_byteMem0Contents, oe, we, byteEnable1, dataIn, address) IS
   BEGIN
      IF (rising_edge(clock)) THEN
         IF (byteEnable1 = '1') THEN
            IF (we = '1') THEN
               s_byteMem1Contents(to_integer(unsigned(address))) <= dataIn(15 DOWNTO 8);
            END IF;
         END IF;
      END IF;
   END PROCESS mem1w;

END platformIndependent;

```
When I compile it with quartus, it gives:
```
Info (276007): RAM logic "main:CIRCUIT_0|RAMCONTENTS_RAM_1:RAM_1|s_byteMem0Contents" is uninferred due to asynchronous read logic
```
According to https://www.intel.com/content/www/us/en/programmable/quartushelp/13.0/mergedProjects/msgs/msgs/iinfer_read_logic_is_asynchronous.htm, quartus should implemented the "asynchronous RAM" with registers. I think this will be a better choice to overcome the chained asynchronous read problem.

dxhisboy, 2022-12-12:
In my opinion, allowing the "gated clock" condition would allow FPGA experiments done in a step-by-step way (connecting the clock pin with a button), which is valuable for educational purposes. @BFH-ktt1

maehne, 2022-12-15:
Using a button as clock source is always risky when implementing it on FPGA due to bouncing. We routinely expose our students to this problem to train them to use proper clocks instead. IMHO, we should strive that the generated HDL corresponds closely to the behaviour implemented for simulation in Logisim-evolution. If there are inconsistencies, it needs to be decided which side is closer to the behaviour of the actually modelled component.

dxhisboy, 2022-12-18:
> Using a button as clock source is always risky when implementing it on FPGA due to bouncing. 
We understand the risk and have a very stable clock button with de-bouncing circuits.

Also, the bouncing can also be solved inside the FPGA with a small filter module.

dxhisboy, 2022-12-18:
> If there are inconsistencies, it needs to be decided which side is closer to the behaviour of the actually modelled component.
The 4 subcycles solution seems totally different with asynchronous RAM. What if I provided an address with IO components?

maehne, 2022-12-20:
@BFH-ktt1: Could you please provide feedback on @dxhisboy additional remarks?

dxhisboy, 2022-12-29:
Also, clock gating is not rare in circuit design.

Our experiment guide told us to invert clock and make registers written in falling edge in the schematic editor of Quartus.

Other website also illustrates similar strategy, e.g. https://www.fpgakey.com/tutorial/section535 .

maehne, 2023-01-19:
In general, Logisim-evolution already allows you to choose the clock edge to be used by a flip-flop or register using the "Trigger" property. Registers provide you with a "WE" input aka. clock enable, which is the recommended strategy when targeting FPGAs with their limited number of dedicated global clock networks that is also described in the reference you provide. For a D flip-flop the clock enable can be easily added by placing a multiplexer in front of its D input, which either passes the new value in case the enable is active or the D flip-flop's current value. Logisim correctly simulates clock gating from an idealised perspective.

Regarding the this particular issue and the observed inconsistency, @BFH-ktt1 is most knowledgeable about the implementation details and therefore can best comment/address your issue.

chuckb, 2023-02-14:
I also demonstrate simulation inconsistency with the RAM module in a video [HackCPU Logisim CPU Testing](https://www.youtube.com/watch?v=UQYPcBNLSWA&list=PLM8YDhk_PWu00p4lqJBMeJtLzYEtE69Y3&index=19). I talk about block RAM configuration settings that might be tweakable to bring the simulation in line with how the hardware behaves, but I have not explored those ideas yet.

BFH-ktt1, 2025-06-01:
Closing this issue as my PR got merged. Please reopen if you find that the problem is not resolved.

closed 2025-06-01 by BFH-ktt1