logisim-evolution/logisim-evolution #1602: Unknown lag behavior

waynezhang2021, 2022-12-12:
[CPU.zip](https://github.com/logisim-evolution/logisim-evolution/files/10208229/CPU.zip)
I'm designing a CPU as the file attached.
I don't know why the software started lagging recently.

maehne, 2022-12-15:
Please be more specific in your error description so that we can more easily reproduce the observed misbehaviour! In general, the GUI of Logisim-evolution can start reacting slowly if the design is very big and simulation is enabled - especially with auto-ticks enabled for the clocks. When I quickly looked at your design, I noticed that you use instead of busses individual signals/ports for each bit. This can make a great impact on simulation performance, because events on these signals need to be processed individually and then also displayed accordingly. My recommendation would be to always group together signals to busses, which conceptually belong together. Where needed, busses can be alway split up using the Splitter component.

I am closing this issue, because it seems more caused by the used design style in `CPU.circ` than exposing a problem in Logisim-evolution. Feel free to reopen if problems persist after you converted your design to use busses where it makes sense.

closed 2022-12-15 by maehne

waynezhang2021, 2022-12-15:
Oh I see and I will change my design a bit.Thanks!