logisim-evolution/logisim-evolution #1604: Timing diagram: Export image in GIF format is causing error message

maehne, 2022-12-13:
If one tries to export the timing diagram in GIF format using `File -> Export Image...`, one gets the following error message:

[image]

Exporting the timing diagram in PNG format works. Interestingly, exporting the schematic in GIF format works.

Logisim-evolution: 3.8.0dev (3e1efb0c)
Java: OpenJDK Runtime Environment Zulu18.32+11-CA (build 18.0.2+9)
OS: Windows 11 Pro 22H2 x86_64

Oren14Dani, 2023-01-03:
Hey there!!!
I would like to grab this one!
Can I have a pin point or some background??? 😎

maehne, 2023-01-03:
@Oren14Dani: Thanks for your willingness to pick up this issue! When you are at it: Would you mind to also have a look at related issues #1605 and #1606?

I am not familiar with this part of the code base and therefore can provide only rough pointers. The timing diagram implementation is located in `src/main/java/com/cburch/logisim/gui/chrono/`. The therein contained `ChronoPanel` has a `PrintHandler` instance, which it overrides probably in an incomplete way. `PrintHandler.java` is located in `src/main/java/com/cburch/logisim/gui/menu/` and may be also worth checking out @kevinawalsh's Logisim-evolution (Holy Cross Edition) fork for a possible fix. @kevinawalsh largely improved this part of Logisim-evolution and his patches might have been incompletely ported back into Logisim-evolution by @BFH-ktt1. 

@BFH-ktt1 and @kevinawalsh: Would you have any additional pointers?