logisim-evolution/logisim-evolution #1610: Support 8 bit and 16 bit floating point formats

maehne, 2022-12-15:
### Discussed in https://github.com/logisim-evolution/logisim-evolution/discussions/1608

<div type='discussions-op-text'>

<sup>Originally posted by **juancolonna** December 14, 2022</sup>
Dear developers, first I'd like to thank you for the amazing work you are doing developing Logisim. Second, I would like to request a new feature if possible.  

I use Logisim as a teaching tool with my students.  I usually explain initial concepts using 8-bit logic, in a simple way.  I often using minifloat (8-bit float) to explain floating number concepts (scale, overflow, etc.).  Floating arithmetic blocks don't have the 8-bit option, eg: addition, multiplication, subtraction, etc., and for me it would be very useful.  Would it be possible to implement 8-bit float operations?  I would be very grateful.

At,</div>

maehne, 2022-12-15:
@juancolonna: I converted your discussion topic to a feature request. PRs in this direction are welcome.

juancolonna, 2022-12-18:
Dear @maehne , thanks for your reply. I don't know how to do a PR. I tried to look the source code of Logisim and I feel a little loss. Is there any documentation to guide me?

At,.

maehne, 2022-12-20:
@juancolonna: GitHub provides good documentation [to get you started](https://docs.github.com/en/get-started/quickstart) on how to use `git` and their service. Check out [Logisim-evolution's developer documentation](https://github.com/logisim-evolution/logisim-evolution/blob/master/docs/developers.md) for specific information on how to contribute to our project.

Regarding the Logisim-evolution source code itself, there's the chapter "JAR libraries" in the "Guide to Being a Logisim User", which you can access from `Help -> User's Guide`. Unfortunately, it's very likely that some of the information given in there is outdated. However, you you should be able to fill up the holes by looking at the source code itself and asking here. For your feature, have a look at the `Fp*.java" files in `src/main/java/com/cburch/logisim/std/arith/`. As I pointed out in discussion #1608:

> Implementation of the floating point arithmetic blocks is currently based on Java's double data type. Smaller floats would need to get converted manually back and forth to this representation (e.g., see [here](https://stackoverflow.com/questions/37105764/how-to-convert-to-from-an-8-bit-float-representation)). One would not only need to adapt these blocks, but also the port and probe components (so that the signal values can be correctly shown when choosing the radix "Float".

In a first step, I would probably start modifying one of the floating point components (e.g., [`FpNegator.java`](https://github.com/logisim-evolution/logisim-evolution/blob/master/src/main/java/com/cburch/logisim/std/arith/FpNegator.java) or [`FpAdder.java`](https://github.com/logisim-evolution/logisim-evolution/blob/master/src/main/java/com/cburch/logisim/std/arith/FpAdder.java)) and try to add the additional bitwidths (8 bit, 16 bit) and modify the conversion of the input values to `double` and the result to the output value to see whether it works. Only then, I would suggest modifying the ports and probes source code to allow direct display of these floating point values.

R3dst0ne, 2022-12-24:
IEEE 754-2019 defines 64, 32 and 16 bit Floating-Point Arithmetic. (the bigger ones are of no interest to LSe.)
I support adding the 16 bit variant.
But to my knowledge, 8-bit Floats are not standardized. Please correct me if I miss something.
If they get added, it is important to document how they work and that they do not follow the IEEE 754 standard.

18001767679, 2023-01-05:
You do not have 8 bit floats.
They are very small and cannot float.
1 bit sign,3 bit exponent,4 bit mantissa?

maehne, 2023-01-10:
Carl Burch's chapter ["Floating-point representation"](http://www.cburch.com/books/float/) from his book ["Computing Systems Organization"](http://www.cburch.com/cso/) might be a base for implementing/documenting this feature.

maehne, 2026-08-21:
Closing this issue as resolved by PRs #1878 and #2425.

closed 2026-08-21 by maehne