logisim-evolution/logisim-evolution #1616: looking for cooperate on verilog converter

quantrpeter, 2022-12-31:
Hi
   My name is Peter, system architect from [www.quantr.foundation](www.quantr.foundation), we are team of 7 in Hong Kong working on RISC-V cpu. We have huge passion on low-level stuff. We want to create a verilog converter, to convert verilog into logisim and simulate it, mainly for education purpose. We have our own verilog parser built by antlr. We have used it to build our own verilog preprocessor [https://www.quantr.foundation/project/?project=Verilog%20Compiler](https://www.quantr.foundation/project/?project=Verilog%20Compiler) and debugger [https://www.youtube.com/watch?v=cPjEUe2uY8I](https://www.youtube.com/watch?v=cPjEUe2uY8I) . That prove we are serious.

   Now we want to find partners, especially on consulting roles.
1. We want to know how to translate verilog into ray logic, only compose by d-latch and gates.
2. We don't know how to import thing to logicism, need some guidance on hacking the logicsim
thanks
Peter

maehne, 2023-01-03:
@quantrpeter: Thanks for your interest in Logisim-evolution as a teaching tool and reaching out to us for enhancing it!

Translating Verilog to a Logisim circuit boils basically down to implementing logic synthesis. IMHO, this is out of scope for Logisim-evolution itself, as there are other open source tools, e.g., Yosys, which implement very well this complex task. What I could imagine is doable is teaching Yosys how to synthesise to Logisim-evolution's built-in libraries using a similar approach as described [here](https://hackaday.io/project/180839-vhdlverilog-to-discrete-logic-flow/log/199967-synthesis-and-technology-mapping). Once one has the gate level netlist, this could be translated to a valid Logsim-evolution `.circ` file, which is "just" simple XML. I am afraid that there is not much documentation available about this format other the Logisim-evolution source code itself.

Are you aware about the "System On a Chip" library, which needs to be explicitly enabled via Project -> Load Library -> Built-in Library...? It already contains a Risc V IM simulator capable of executing ELF files. Unfortunately, there is not much documentation available, but feel free to ask here for help. @BFH-ktt1 implemented it and can assist you. We are also open for improvements in this area.

Logisim-evolution has also the capability to co-simulate VHDL entities and Logisim-evolution circuits using ModelSIM/Questa Sim. Extending this to also Verilog modules should be doable with reasonable effort. It should be also feasible to support in addition open-source HDL simulators such as ghdl, iverilog, and verilator. Work in this direction would be welcome.

quantrpeter, 2023-01-03:
Thanks @maehne , I and my team still want to build the verilog compiler to convert verilog into logisim, discovery comes from journey. Is there any java library can generate that .circ file? Or I need to hack it?

maehne, 2023-01-04:
I understand your motivation and the decision is of course up to you. I just wanted to make you aware of the existing open source ecosystem and point out a way how to efficiently combine them. There is also a lot to be learned by doing this. Especially in the EDA domain combining different tools from different vendors to IC design/verification flows is common and mastering this is an important part of the job.

Still, I would suggest that you try to reuse established tools/algorithms where it make sense to achieve a satisfying feature set and performance. Both, `yosys` and `ghdl`, implement logic synthesis from Verilog and VHDL respectively and are currently the gold standard in the open source world. Logisim-evolution contains limited logic synthesis capability from truth tables and Boolean equations and is capable of creating optimised combinatoric sub-circuits from them. The implementation for this "Analyze" feature can be found in `src/main/java/com/cburch/logisim/analyze/`. The circuit file handling is implemented in `src/main/java/com/cburch/logisim/file/`.

Currently, the Logisim-evolution source code is built in a monolithic way to a single JAR. Making it more modular and reusable for external tools may be interesting. Efforts in this direction from new contributors would be certainly considered for integration as would be any contributions to the documentation of the API and development of unit tests. Logisim-evolution already allows loading of JAR libraries, but the state of this API is a bit unknown to us. It hasn't been used much in the past years by the main contributors. You may also try to integrate your new features directly into Logisim-evolution and submit a PR once it is in a stable and usable state for external users.

quantrpeter, 2023-02-04:
Finally collected enough resources to start, we start to build a java library to export the data structure in java to .circ file.

quantrpeter, 2023-02-04:
[image]

What is the reason we need to define the <toolbar> in the .circ file. The toolbar should be depend on the logisim version, not the project itself.

R3dst0ne, 2023-02-04:
I can't answer your question, but I noticed that the arrow in your picture is wrong.
Here is a corrected version:
[image]

quantrpeter, 2023-02-05:
thanks, how to reset the toolbar and the tree on the left hand size to full feature? i mean having all buttons?

R3dst0ne, 2023-02-05:
The toolbar (, mouse and simulation) can be reset in `Project>>Options...>>Reset`.
Libraries can be loaded in `Project>>Load Library`.
Those changes are specific to the current project. If you want them to be the default on startup, you have to create a template file containing these changes.

closed 2026-10-21 by BFH-ktt1