logisim-evolution/logisim-evolution #1619: Какой идиот решил для DFF использовать клок снизу?

ogamespec, 2023-01-03:
Привет :)

Общемировая практика использования клока в обозначении DFF - сверху элемента.

Приходится крутить петли, чтобы развести сигналы.

Исправьте, пожалуйста, спасибо.

BFH-ktt1, 2023-01-03:
@ogamespec : Could you please ask your question in English? I do not speak/read your language unfortunately.

maehne, 2023-01-03:
@ogamespec: I agree with @BFH-ktt1 that you should strive for asking your questions in the common language used by a software projects' contributors. DeepL translates your question to:

> Hi :)
>
> It's common practice to use a clock in the DFF designation - on top of the element.
>
> I have to twist the loops to separate the signals.
>
> Please correct this, thank you.

If it is correct, I am afraid that we cannot address your issue: For backward-compatibility reasons, pin positions in symbols cannot be changed. Otherwise, we would break a lot of existing Logisim circuit schematics. Also, the current D flip-flop symbol is following [common conventions](https://en.wikipedia.org/wiki/Electronic_symbol#Flip-flops). If you are unhappy with the pin placement, you can easily create a sub-circuit as a wrapper to Logisim's D flip-flop and adjust its symbol. Another way to avoid crossing other wires is using a Tunnel primitive from the Wiring library.

I will close this issue as not planned. Feel free to reopen if you have additional arguments or would like to provide a PR for your suggested enhancement (in a backward compatible way).

closed 2023-01-03 by maehne