logisim-evolution/logisim-evolution #1620: gradlew build not working

quantrpeter, 2023-01-04:
Dear All, please help

```
/Users/peter/workspace/logisim-evolution>./gradlew build
<<======-------> 47% EXECUTING [8s]
> Task :checkstyleMain FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':checkstyleMain'.
> Expected file collection to contain exactly one file, however, it contains no files.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 11s
13 actionable tasks: 9 executed, 4 up-to-date
```

quantrpeter, 2023-01-04:
[image]

[image]

thanks

quantrpeter, 2023-01-04:
more hints: gradlew.bat checkstyleMain --stacktrace --info

```
> Task :classes UP-TO-DATE
Skipping task ':classes' as it has no actions.
:classes (Thread[#2251,Execution worker Thread 16,5,main]) completed. Took 0.0 secs.
Resolve mutations for :checkstyleMain (Thread[#2241,Execution worker Thread 6,5,main]) started.
Resolve mutations for :checkstyleMain (Thread[#2241,Execution worker Thread 6,5,main]) completed. Took 0.0 secs.
producer locations for task group 0 (Thread[#2247,Execution worker Thread 12,5,main]) started.
:checkstyleMain (Thread[#2246,Execution worker Thread 11,5,main]) started.
producer locations for task group 0 (Thread[#2247,Execution worker Thread 12,5,main]) completed. Took 0.0 secs.

> Task :checkstyleMain FAILED
:checkstyleMain (Thread[#2246,Execution worker Thread 11,5,main]) completed. Took 0.043 secs.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':checkstyleMain'.
> Expected file collection to contain exactly one file, however, it contains no files.

* Try:
> Run with --debug option to get more log output.
> Run with --scan to get full insights.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':checkstyleMain'.
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:38)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:204)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:199)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:53)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:73)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
        at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:69)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:327)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:314)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:307)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:293)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:420)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:342)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
        at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:48)
Caused by: java.lang.IllegalStateException: Expected file collection to contain exactly one file, however, it contains no files.
        at org.gradle.api.internal.file.AbstractFileCollection.getSingleFile(AbstractFileCollection.java:164)
        at org.gradle.api.internal.resources.FileCollectionBackedArchiveTextResource$1.createDelegate(FileCollectionBackedArchiveTextResource.java:43)
        at org.gradle.api.internal.file.collections.LazilyInitializedFileCollection.visitChildren(LazilyInitializedFileCollection.java:33)
        at org.gradle.api.internal.file.CompositeFileCollection.visitContents(CompositeFileCollection.java:119)
        at org.gradle.api.internal.file.AbstractFileCollection.visitStructure(AbstractFileCollection.java:375)
        at org.gradle.api.internal.file.CompositeFileCollection.lambda$visitContents$0(CompositeFileCollection.java:119)
        at org.gradle.api.internal.file.collections.UnpackingVisitor.add(UnpackingVisitor.java:64)
        at org.gradle.api.internal.file.collections.UnpackingVisitor.add(UnpackingVisitor.java:89)
        at org.gradle.api.internal.file.DefaultFileCollectionFactory$ResolvingFileCollection.visitChildren(DefaultFileCollectionFactory.java:333)
        at org.gradle.api.internal.file.CompositeFileCollection.visitContents(CompositeFileCollection.java:119)
        at org.gradle.api.internal.file.AbstractFileCollection.visitStructure(AbstractFileCollection.java:375)
        at org.gradle.api.internal.file.CompositeFileCollection.lambda$visitContents$0(CompositeFileCollection.java:119)
[... cut ...]
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:420)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:342)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
        at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:48)

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
5 actionable tasks: 1 executed, 4 up-to-date
Watched directory hierarchies: [C:\workspace\logisim-evolution]
```

maehne, 2023-01-04:
For `./gradlew createMsi`, you need to install the [WiX tools](https://wixtoolset.org/docs/wix3/), as stated in the Gradle output.

quantrpeter, 2023-01-04:
resolved, i changed these

```
    val archive = configurations.checkstyle.get().resolve().filter {
      name.startsWith("checkstyle")
    }
```

to 

```
    val archive = configurations.checkstyle.get().resolve().filter {
      it.name.startsWith("checkstyle")
    }
```

MarcinOrlowski, 2023-01-04:
As a side note I'd recommend to skip checkstyle checks during unless you want to fix them. The reason is that LSe code is still causing problems there:

```bash
$ gradlew build -x checkstyleMain -x checkstyleTest
```

maehne, 2023-01-05:
Reopen issue since it is not yet fixed in branch `develop`.

BFH-ktt1, 2023-01-11:
@maehne : Can this issue be closed due to merge of #1624 ?

maehne, 2023-01-12:
@BFH-ktt1: Yes, you can close this issue as fixed.

maehne, 2023-01-13:
Fixed with PR #1624.

closed 2023-01-13 by maehne