logisim-evolution/logisim-evolution #1622: Does not save pin layout

18001767679, 2023-01-04:
If you only change the layout of the pins,and do not delete the rectangle,the changes will not be saved.

18001767679, 2023-01-07:
It's normal now.Maybe its just that time it got a bug

18001767679, 2023-01-07:
I want to post a little question offtopic here:How do i set poke behaviour for VHDL?My computer design is ported to vhdl now but now its memory is inaccessible.

maehne, 2023-01-10:
@18001767679: Thanks for the feedback that the issue is resolved on your side.

Regarding:

> I want to post a little question offtopic here:My vhdl simulation only runs at 45 fps,as opposed to my cpu which runs as 300 fps so integrating it into vhdl was a bad idea how do i fix

This question is unrelated to this issue. It's more fitting to create a [discussion thread](https://github.com/logisim-evolution/logisim-evolution/discussions) for it. However, co-simulation comes always with a cost/performance penalty. After all, both simulators need to get synchronised each time a signal changes at the interface -- even if it is only the clock signal.

I don't have any experience how big is the performance penalty for Logisim <-> QuestaSim, as I have hardly used the feature myself. Once we start teaching our students VHDL, we start to use directly the vendor EDA tools.

I will close this issue as resolved.

closed 2023-01-10 by maehne