logisim-evolution/logisim-evolution #1635: Bad mouse

18001767679, 2023-01-12:
The joystick has blind spots(connect itto a rgb video and turn on clock and see)
Also It has 5 bits precision and I need 8

maehne, 2023-01-16:
I modified the title of the issue, as Logisim-evolution offers no mouse component. I cannot reproduce your issue regarding blind spots of the joystick. For me, it generates all possible 2**5=32 values on both axes, x and y. Though, it may jump values if you have zoomed too much out of your schematic. However, we cannot do much about that, as the mouse attached to the computer also has a limited resolution.

I don't know the history of this component, but it might well be that its maximum resolution of 32 values per direction is on purpose due to its small size. Resolution could be probably increased with a reasonable effort, but it is not a priority for us. PRs in this direction are welcome if backward compatibility is ensured.

BFH-ktt1, 2023-01-17:
I'm sorry, but this kind of request I find a bit awkward: "It has 5 bit and I need 8".... I'm happy to aid, but I do not see how I can satisfy the requests of each and every person in the world. I also would like that LE would ignite my fireworks on the first of August (which is an important date in Switzerland), but it is probably only me who likes that feature. Any PR's are welcome, but up to that moment I close this issue as "won't fix".

closed 2023-01-17 by BFH-ktt1