logisim-evolution/logisim-evolution #1639: Touch screens

18001767679, 2023-01-16:
It would be nice if there were touch screens

maehne, 2023-01-16:
Please, be more specific regarding the interface and expected behaviour! How should the user interact with that component? Which real-world component(s) it should try to resemble in behaviour? What use cases should it fulfill? When you suggest new features without proposing an implementation in form of a PR, you should at least motivate your request and provide enough detail that someone may find interest in your proposal to step in for the implementation. 

For us such a feature is of (very) low interest: Yes, you can build virtual prototypes (VPs) of complete embedded systems in Logisim-evolution and execute compiled programs. However, the achievable performance is limited by several factors making real time interaction with those VPs cumbersome once a certain limit of complexity is passed. That's more the realm of professional tools and system modeling frameworks like SystemC.

PRs in this direction are welcome if motivated by answering the questions above and providing enough documentation to be usable by third parties. If no one shows interest in the coming weeks, I will close it as stale.

BFH-ktt1, 2023-01-17:
I agree with @maehne , and close this issue as "won't implement". Please make a new issue with detailed information.

closed 2023-01-17 by BFH-ktt1