logisim-evolution/logisim-evolution #1642: Controlled Buffer component’s output is "E" when input is U and buffer is enabled

prlaba, 2023-01-17:
I consider this a bug.

If a Controlled Buffer component's input pin is floating (U), it generates an error (E) on its output pin if the buffer is enabled.

The correct behavior would be to generate U on the buffer's output pin when its input pin is U and the buffer is enabled.  In other words, the buffer should always pass the state of its input pin (0, 1 or U) to its output pin when enabled.

The only time the buffer should generate E on its output pin is when its Enable pin is disconnected or its input pin is E.

This bug makes it impossible to create a bi-directional buffer using two controlled buffers with their outputs tied to each other's inputs (like the 74245 8-bit bus transceiver).

Interestingly, the latest version of L-E (8.03) includes a custom 74245 IC that works correctly (both the A and B sides can both be floating, with no errors when enabled). But I can't build that same IC using L-E's Controlled Buffer components because of this bug.

maehne, 2023-01-19:
@prlaba: I agree with your assessment. Interestingly, the documentation of the controlled buffer contradicts itself:

[image]

The text describes the behaviour, which you expect, but the truth table does not reflect that and specifies the observed problematic behaviour. The buffer component propagates the `U`, which is also specified in its documentation.

davidhutchens, 2023-04-24:
With the merge of #1726, this issue should be fixed in the develop branch. I am closing this as completed. 

@prlaba: You can build from develop, or download from our Nightly. Please reopen if problems remain.

closed 2023-04-24 by davidhutchens

prlaba, 2023-04-24:
Thanks, I’ll text and confirm.

prlaba, 2023-04-24:
“test”

commit 13e5b778 (2023-04-11): Merge pull request #1704 from seanpquinn/controlled_buffer_U

[ControlledBuffer] fix logic, closes #1642
commit f81cb2c4 (2023-04-11): [ControlledBuffer] fix logic, closes #1642