logisim-evolution/logisim-evolution #1647: Fixing the Snapcraft package

anrouxel, 2023-01-19:
Hello @R3dst0ne,

I have identified some errors in the snapcraft.yaml file that is used to create the Snap package for Logisim-evolution. The specific issues are as follows:
- The source URL for the project is incorrect. It should be `https://github.com/logisim-evolution/logisim-evolution.git` instead of `https://github.com/MarcinOrlowski/logisim-evolution.git`.
- The branch that is checked out during the build process is incorrect. It should be specified as the correct branch for the project.

Here is the file your snapcraft.yaml file :
```
#
# Control file to build SNAP package of LSe
# https://github.com/logisim-evolution/logisim-evolution
# https://snapcraft.io/
#
# Marcin Orlowski
#

name: logisim-evolution
base: core22
confinement: strict
icon: support/jpackage/linux/logisim-icon-128.png
license: GPL-3.0
summary: Digital logic design tool and simulator
description: |
 Logisim-evolution is educational software for designing and simulating digital logic circuits.
 Logisim-evolution is free, open-source, and cross-platform.
 Project highlights:
  * easy to use circuit designer,
  * logic circuit simulations,
  * chronogram (to see the evolution of signals in your circuit),
  * electronic board integration (schematics can be simulated on real hardware),
  * VHDL components (components behavior can be specified in VHDL!),
  * TCL/TK console (interfaces between the circuit and the user),
  * huge library of components (LEDs, TTLs, switches, SoCs),
  * supports multiple languages,
  * and more!
grade: stable

adopt-info: logisim-evolution

architectures:
  - build-on: amd64

layout:
  /etc/fonts:
    bind: $SNAP/etc/fonts
  /usr/share/fonts:
    bind: $SNAP/usr/share/fonts
  /var/cache/fontconfig:
    bind: $SNAP_DATA/var/cache/fontconfig

apps:
  logisim-evolution:
    extensions: [gnome]
    command: usr/lib/jvm/java-17-openjdk-$CRAFT_TARGET_ARCH/bin/java -jar $SNAP/logisim-evolution/logisim-evolution.jar
    environment:
      _JAVA_OPTIONS: "-Duser.home=$SNAP_USER_DATA"
      XDG_DATA_HOME: $SNAP/usr/share
      FONTCONFIG_PATH: $SNAP/etc/fonts/conf.d
      FONTCONFIG_FILE: $SNAP/etc/fonts/fonts.conf
    plugs:
      - desktop
      - desktop-legacy
      - gsettings
      - home
      - opengl
      - unity7
      - wayland
      - x11

parts:
  logisim-evolution:
    plugin: nil
    source: https://github.com/MarcinOrlowski/logisim-evolution.git
    source-type: git
    override-pull: |
      craftctl default
      git checkout snap-packages-test-1
      git status
      LSE_VERSION="$(cat gradle.properties | grep '^version\s*=.*' | awk '{print $3}')"
      craftctl set version="${LSE_VERSION}"
    override-build: |
      bash gradlew "$@" shadowJar && \
      mkdir -p "$CRAFT_PART_INSTALL"/logisim-evolution && \
        ls -ld build/libs/*
        mv build/libs/logisim-evolution-*-all.jar "$CRAFT_PART_INSTALL"/logisim-evolution/logisim-evolution.jar
    build-packages:
      - git
      - openjdk-17-jdk
      - libfontconfig1-dev
    stage-packages:
      - openjdk-17-jre
      - fonts-freefont-ttf
      - fonts-arphic-uming
      - fontconfig-config
```

Best Regards

maehne, 2023-01-24:
@anrouxel: Thanks for pointing out these issues!

@R3dst0ne and @MarcinOrlowski: Could you please have a look?

R3dst0ne, 2023-01-25:
As mentioned in #1614 I sadly do not have any experience with Snap's or configuring build processes.

@anrouxel thank you for looking into this.

MarcinOrlowski, 2023-02-21:
Thanks for taking a look @anrouxel I will address the issues you detected.

MarcinOrlowski, 2023-03-02:
@maehne I pushed 3.8.0 to snap store now, yet I keep this ticket open for a while as I had some issues with builds based on `core22`, so 3.8.0 is based on `core20`, Also, I'd like to have nightly builder to also create `*.snap` for downloads (and/or perhaps push to `latest/edge` to close this ticket.

maehne, 2023-03-03:
@MarcinOrlowski: Thanks!

**TODO:**

- [ ] Build snap based on `core22` instead of `core20`
- [ ] Build snaps using a nightly builder for download
- [ ] Push new snaps to `latest/edge` channel automatically

R3dst0ne, 2023-03-10:
You updated the name, but it is still not the official one.
`Logisim-evolution` is the correct name of this software.
The package name is `logisim-evolution` and the name of the organization is `Logisim-Evolution`.
Please fix this if possible.

maehne, 2025-09-05:
@MarcinOrlowski: Could you maybe have again a look to this Snap packaging issue?