logisim-evolution/logisim-evolution #1650: How can I use quartus lite without a license within logisim

spidertyler2005, 2023-01-24:
Hey, I am just wondering how I would do this. I am a complete noob and only just installed quartus lite. I am having trouble using it with logisim since I am using the free version which does not have a license.

Thanks in advance.

R3dst0ne, 2023-03-26:
@spidertyler2005
Did @maehne answerer help you resolve your issue?
If yes, could you please close this issue.
Otherwise, please provide information where you got stuck.

spidertyler2005, 2023-12-01:
I cant honestly remember if i got this working. My hdd corupted some time after this point so I had to reinstall windows for the 5th time in the past 2 years.

Im gonna try to close this issue tho because i dont want to clog up the issues page any further.

closed 2023-12-01 by spidertyler2005