logisim-evolution/logisim-evolution #1651: New app icon for macOS to fit with Apple's Human Interface Guidelines

TheWeirdSquid, 2023-01-24:
I've created a new app icon for Logisim-evolution that is based off the original, but fits the style of the new macOS app icons. I've attached the .icns file as well as the blend file I used.
[logisim.icns.zip](https://github.com/logisim-evolution/logisim-evolution/files/10492708/logisim.icns.zip)
[logism evolution logo.blend.zip](https://github.com/logisim-evolution/logisim-evolution/files/10492710/logism.evolution.logo.blend.zip)

[image]

If someone would let me know what I need to change to implement this, I will make my own fork and submit a PR.

TheWeirdSquid, 2023-01-24:
Here's an alternative version that I feel fits the theme of the program better.
[logisim circuit.icns.zip](https://github.com/logisim-evolution/logisim-evolution/files/10494711/logisim.circuit.icns.zip)
[logism evolution logo circuit.blend.zip](https://github.com/logisim-evolution/logisim-evolution/files/10494712/logism.evolution.logo.circuit.blend.zip)
[image]

R3dst0ne, 2023-01-25:
> If someone would let me know what I need to change to implement this, I will make my own fork and submit a PR.

I do not know what you have to change, but maybe you are able to figure it out by looking into PR #904.

---

I personally prefer the first icon.
IMHO, the PCB background of the second one suggest a wrong idea about what LSe is for. Since LSe does only feature digital simulation, I think the Icon shouldn't feature elements that could leave the impression that it supports analog as well.

maehne, 2023-01-27:
@TheWeirdSquid: Thanks for proposing this new application icon for Logisim-evolution. I concur with @R3dst0ne that I prefer the first version. Would you be able to provide also a suitable icon for the `.circ` files? The icon(s) should go into [`support/jpackage/macos/`](https://github.com/logisim-evolution/logisim-evolution/tree/master/support/jpackage/macos) for them to get picked up by `jpackage` when it is called by the [`gradle createApp` task](https://github.com/logisim-evolution/logisim-evolution/blob/0f2bb014611ce7520e27cce70242ca1438eede25/build.gradle.kts#L380). It would be nice if you could prepare a PR, so that you get also full credit for this contribution.

TheWeirdSquid, 2023-02-06:
Absolutely, thanks for the help. I'll make a PR as soon as I've got the time.

TheWeirdSquid, 2023-02-07:
> @TheWeirdSquid: Thanks for proposing this new application icon for Logisim-evolution. I concur with @R3dst0ne that I prefer the first version. Would you be able to provide also a suitable icon for [...]

In regards to the .circ files, I believe MacOS will give them the icon of whatever app they open in. If the app icon is updated, the .circ files should also update. I'm not positive about this, but I've been playing around with app icons for a bit and this seems to be how they work.

davidhutchens, 2023-02-07:
@TheWeirdSquid: macOS does not use the app icon as the document icon. In fact, a given app can define multiple document icons. In Logisim-evolution, there are two icon files in `support/jpackage/macos`. One is the app icon, the other is the document icon (which contains circ in the name). They happen to be two copies of the same file. But they need not be, and probably shouldn't be.

You can replace the app icon and leave the document icon as it was. That way, they would be related, but no longer identical.

davidhutchens, 2023-02-10:
I recommend that you update the app icon and leave the document icon as it is. We are using that icon image on the Windows and Linux builds also.

Please make sure the images are crisp at their given sizes. The ones in your example above look fuzzy to me.

TheWeirdSquid, 2023-02-13:
Alright no problem. As for image crispness, the .icns file contains a number of different resolutions of the image. I've turned off the emission effect for smaller resolutions to maintain crispness. I agree, the image I sent at the top of the thread does look blurry. I think this is due to how Github decided to scale it, but I have not had any issues while testing the icon on the computer.

TheWeirdSquid, 2023-02-13:
I've submitted a PR: https://github.com/logisim-evolution/logisim-evolution/pull/1668

TheWeirdSquid, 2023-02-16:
I've submitted a new PR: https://github.com/logisim-evolution/logisim-evolution/pull/1672

davidhutchens, 2023-09-05:
This issue was completed with the merger of #1672.

closed 2023-09-05 by davidhutchens

JDMinze, 2024-10-31:
> Here's an alternative version that I feel fits the theme of the program better. [logisim [...]

Technically, apple icons are squircles, not just squares with rounded corners and have a shadow cast. You can find the Apple design template here: https://developer.apple.com/design/resources/#macos-apps
Here's a modified version that has the correct shape and shadow:
[image]
The center icon probably should have been resized to follow Apple's gridline suggestions, but I couldn't be bothered, and is optional. Here's a Figma link if you would rather see it that way: https://www.figma.com/design/KYKAihPuwHO3NK9Bc4vq6H/Logisim-Icon?t=BjzmMP21g7jGSBjS-1