logisim-evolution/logisim-evolution #1655: I am not able to export simulatio as text file.

Juan-Gg, 2023-01-28:
In the Documentation for logging it mentions the following:

> "Buttons: There are three buttons at the bottom of our window.
Load file Loads and displays the result from a previously saved file. This does not allow to set the simulation.
Export... Exports the data into a text file. You will need to specify the file name in the select window. ".

I cannot find those buttons in the Timing Diagram Window. I want to carry out a simulation and export register values as a .txt.

Thanks in advance.
[image]

maehne, 2023-01-30:
@Juan-Gg: Thanks for raising the issue! For sure, the documentation is outdated. Even some images are currently missing:

[image]

@BFH-ktt1 merged improvements to the chronogram from @kevinawalsh's [holy cross fork](https://github.com/kevinawalsh/logisim-evolution). Its documentation in the Logging chapter is even more outdated. :-( So, I am not sure that the described functionality is even still present in the code base.

@mbaillif: I guess that you updated the documentation. Could you maybe have a look?

Juan-Gg, 2023-01-30:
I figured. I had a quick look at the source code and didn't see anything related to logging on the chronograph.
The logging functionality was quite handy to debug large circuits across hours of simulation, I used to use it on the original logisim. I hope it gets reintroduced eventually. Also the chronograph loads down the simulation quite a bit in my case.  (I made do with some rams and a counter to log the data I needed.)
Thanks.

hewzhew, 2026-08-20:
I traced this through both the current code and the history of the chronogram rewrite.

The old UI should not simply be made visible again. In Kevin Walsh's 2018 commit [`784c3d9f4`](https://github.com/kevinawalsh/logisim-evolution/commit/784c3d9f4f8ed9742d2f6826e0b637ebd3cbfa41), the Table and File tabs were deliberately hidden as broken, with this note:

> rethink file writer in light of signal rewriting

That concern still applies. The current `LogThread` only appends rows, while coarse clocked recording can call `Model.replaceWithNewValues()` and rewrite the most recent waveform interval. Re-enabling `FilePanel` as-is could therefore silently leave stale values in the exported file.

The documented **Load file** / **Export...** buttons came from an older, separate chronogram data model. That model and its parser were removed during the rewrite; the old format also required a signal named `sysclk`, whereas the current timing diagram supports arbitrary clock sources. Restoring loading is consequently a separate and substantially larger task.

I see two safe scopes:

1. **Bounded first step (my preference):** Add **Export text...** to the current Timing Diagram and export a UTF-8 TSV snapshot of the retained history. Each row could contain `time_ns`, `duration_ns`, and the selected signals in their current order and radix. The documentation would explicitly explain the history limit and the existing **Store unlimited history** option. This is testable without rebuilding the old import model.
2. **Continuous logging:** Reintroduce a File/logging UI, but drive it from immutable/finalized simulation snapshots rather than the mutable waveform tail. This better serves multi-hour simulations, but it needs a new event/commit contract and a decision about sampling semantics in step, real-time, and coarse/fine clocked modes.

For the first scope I would cover transitions and durations, bus radix formatting, unknown/error values, and truncated-history boundaries. I would leave **Load file** and continuous streaming for follow-ups.

Would the bounded text-export step be an acceptable first PR, or would you prefer the continuous logging contract to be addressed first?

Juan-Gg, 2026-08-20:
@hewzhew I am not an active user of logisim at the moment, so please proceed as you prefer. The first option seems a reasonable course of action.
Thanks.

hewzhew, 2026-08-20:
Thanks, @Juan-Gg.

I have now worked through the exact bounded contract for option 1. Before opening an implementation PR, I propose:

- Add a dedicated **Export text...** action to the current Timing Diagram.
- Export all signals present in the diagram, in their current display order and radix, as UTF-8 TSV with `time_ns`, `duration_ns`, and one column per signal.
- Emit one row for each maximal interval between the union of all signal transition boundaries. Times are absolute nanoseconds and the model end time is exclusive.
- Export only retained history. Data discarded by the history limit is not reconstructed, and `-` represents a signal that was added later and has no value in an earlier retained interval.
- Capture an immutable snapshot before writing so concurrent simulation updates cannot produce internally inconsistent rows. Fields containing tabs, quotes, or line breaks will use conventional delimited-text quoting.
- Default to `.tsv` while accepting `.txt`, update the obsolete documentation, and test staggered transitions, exact boundaries, history truncation, late-added signals, radix/unknown/error formatting, and quoted labels.

I would keep **Load file** and continuous streaming out of this first PR.

@maehne, @BFH-ktt1: Would this bounded contract be acceptable for the first implementation?

maehne, 2026-09-24:
@hewzhew: Sorry for providing so late feedback on your proposal. IMHO, saving digital waveforms in TSV format is ineffient a with limited usefulness. Digital signals typically not change at each time step. Therefore, digital trace file formats such as [VCD](https://en.wikipedia.org/wiki/Value_change_dump), [FST](https://blog.timhutt.co.uk/fst_spec/), and [GHW](https://ghdl.github.io/ghdl/ghw/index.html) only record the events, i.e., time when a signal takes a new value. These three formats are supported by open source viewers, e.g., [surfer](https://surfer-project.org), [Vaporview](https://github.com/Lramseyer/vaporview), and [GTKWave](https://gtkwave.sourceforge.net). For short waveforms, also [WaveDrom](https://wavedrom.com) might be an option. Personally, I think that offering exporting waveforms to either VCD or FST is the better way forward. The proposed TSV format would be more cumbersome to handle by our users without any readily available tooling.

If the TSV format should be retained, I think it is sufficient to have one column with the timestamp when the signals got sampled followed by the column with the current values of the traced signals. I am not sure what purpose `duration_ns` would serve.