logisim-evolution/logisim-evolution #1661: Version 3.8.0 still not released as snap

alculete, 2023-02-06:
Version 3.8.0 still not released as snap on stable channel. If possible, please update this channel

R3dst0ne, 2023-02-07:
This is a known issue.
See #1534 and #1614.
#1647 describes view things that need to be addressed to fix this. 

I'd close this issue as duplicate.

alculete, 2023-02-07:
Post the error you are having on [snapcraft forum](https://forum.snapcraft.io/)  they will help

maehne, 2023-02-10:
@alculete: Do you have experience with creating snap packages and would be willing to help resolve this issue? We have limited ressources and need to prioritize. Typically this means resolving first pain points of our students and ourselves. Since we use it always in combination with FPGA synthesis tool chains, package formats that impose sandboxing have less priority for us.

alculete, 2023-02-10:
Unfortunately no. I think you can ask snapcrafters for help or maybe ask them to take control of your snap releases

MarcinOrlowski, 2023-03-02:
v3.8.0 is now in snap store.

closed 2023-03-02 by MarcinOrlowski

alculete, 2023-03-05:
Hi @MarcinOrlowski  the icon is still not available on the app menu. Still need to go through the terminal to open it