logisim-evolution/logisim-evolution #1664: Obscure error message when trying to open a .circ file containing umlauts

maehne, 2023-02-08:
If a .circ file contains umlauts (e.g. due to automatic renaming in the LMS), Logisim-evolution fails to open the file when double-clicking on it in the Windows Explorer:

[image]

The file name was `Rütti.circ`

shri-acha, 2025-07-14:
https://github.com/logisim-evolution/logisim-evolution/blob/686d6afdffe2bccb9cbc5c89ef31a44973db646f/src/main/java/com/cburch/logisim/gui/start/Startup.java#L502C2-L511C4

I think the filename is taken in with default encoding here, the encoding could be changed to UTF-8. 
Could I help out?

maehne, 2025-07-18:
Thanks for the suggestion @shri-acha. Would you also be able to create a PR with a fix that we can test?