logisim-evolution/logisim-evolution #1666: Synthesized RAM output 0

chuckb, 2023-02-09:
I have a simple design with a counter feeding address 0 of a 256x8 RAM. Every clock tick, write enable is brought high to update RAM with a new value. I examine the signals from the counter and RAM data output. When simulated, data output from RAM is one behind count from counter (as expected). When synthesizing, all RAM data output is low. Board = Alchitry AU with IO hat (using VHDL). Vivado =  v2020.2 (64-bit). Logisim = 3.9.0dev. Any suggestions?

BFH-ktt1, 2023-02-16:
@chuckb : Thank you for reporting this issue, however the problem is in the simulation with logisim:
As you connected the clock to both the clock input and the write-enable, the counter should not write anything in the RAM component, as the clock is "0" at the rising edge (or better set we have a setup-and-hold issue). The VHDL-version does the correct thing, the logisim simulation not. Tying the WE to one solves the problem.
Hope this explains.

maehne, 2023-03-07:
Closing due to inactivity after @BFH-ktt1's reply.

closed 2023-03-07 by maehne