logisim-evolution/logisim-evolution #1667: Frequency questions

chuckb, 2023-02-11:
I have an Alchitry Au board. Board design is set to 100Mhz clock, which matches the HW clock on the board. The greatest synthesis frequency I can pick is 2 Mhz. Why is that?

Next, at 2Mhz, when I synthesize and load the FPGA, I output the clock. My scope says the clock is running at 1Mhz. Why might that be?

chuckb, 2023-02-15:
I altered code locally to add more frequencies. That worked ok...I thought that there might be synthesis register/counter sizing that might be impacted, but that appeared not to be the case. However, it's not necessary.

One can change the divider independent of setting the frequency (did not occur to me until I did it), which precludes needing to add frequencies. I might argue more frequencies would be nice.

But in any case, the maximum frequency I can get out of my board is 12.5 Mhz (using Logisim):
- 100Mhz system clock
- Set divider to 1.
- Frequency then shows 50Mhz (for some reason...not clear why yet)
- Measured frequency on scope shows 12.5Mhz

So is this behavior expected? Why is frequency divided by 2 and then divided by 4?

BFH-ktt1, 2023-02-16:
@chuckb : The explanation to your measurements is following, the frequency that you select is the "tick-frequency". the clock component has (minimal, can be altered in the parameters of the clock component) one tick high and one tick low, meaning a division by two of the "tick-frequency". Hope this explains.

chuckb, 2023-03-05:
@BFH-ktt1 Why the additional division by 4? I noticed a clock tree vector. Is this division to accomodate clock phasing?

chuckb, 2023-03-05:
If I wanted to step up the global clock freq using a vendor specific primitive (for Xilinx, think MMCME2_BASE or PLLE2_BASE), and feed that to the LogisimClockComponent, is there an easy way you can think of to insert this into the application in a plugable way?

BFH-ktt1, 2023-03-14:
> @BFH-ktt1 Why the additional division by 4? I noticed a clock tree vector. Is this division to accomodate clock phasing?

Some of the components are implemented as multi-cycle components, hence need some clock cycles to "emulate" a single-cycle component.

BFH-ktt1, 2023-03-14:
> If I wanted to step up the global clock freq using a vendor specific primitive (for Xilinx, think MMCME2_BASE or PLLE2_BASE), and feed that to the LogisimClockComponent, is there an easy way you can think of to insert this into the application in a plugable way?

This should be doable, as the clock components is inserted on top-level, and hence one could think of adding this for AMD-FPGA's and Intel-FPGA's. Unfortunately I am quite busy this year and will not have time soon to make a PR on this, so it will be on a low priority.

davidhutchens, 2025-10-19:
@BFH-ktt1: was this issue completed with the merge of #1702? #1712 seems to be related to documentation and is still open. Is there more?