logisim-evolution/logisim-evolution #1669: Module to interconnect with other applications

wmurbina, 2023-02-14:
My question is, I could create a module that allows to interconnect with other applications for example with labview.
It could be by serial interface or TCP sockect.
In this video you can see how the Proteus simulation interacts with labview. https://www.youtube.com/watch?v=KbpBKn2fYdU

Thank you.

Emil-DV, 2023-05-22:
I'd like to use Logisim as a UI and drive values from another app.

maehne, 2023-06-06:
IPC with other programs via sockets is already used in Logisim-evolution, as demonstrated by the TCL components and implemented VHDL co-simulation with Questa Sim. However, you'll need to implement your own custom components in Java. You can use the code below `src/main/java/com/cburch/logisim/std/tcl` as a starting point. Unfortunately, documentation on this part of Logisim-evolution is scarce. It was contributed by the people from @reds-heig, which may be able to answer some of your questions, but haven't been very active lately. These parts of Logisim-evolution has seen little contributions in the past years, so that we are not fully sure about their functional state.

Due to our limited resources, such enhancements are low priority for us. PRs in this direction are welcome.