logisim-evolution/logisim-evolution #1673: open file dialog unable to see any file

quantrpeter, 2023-02-18:
i am sure i have file in my desktop but unable to see it. thanks

[image]

quantrpeter, 2023-02-18:
i am in mac, if i run it by command line, then i see the files. If I double click to launch it, it can't

davidhutchens, 2023-02-18:
I can confirm there is some strange behavior on macOS 10.15.7. In macOS, applications can be given permission to access certain folders such as the Desktop. By default, applications do not have access.

At some point in the past, I gave LE permission to see my desktop. Even so, if I try to use the open panel to access a file on my desktop, no files are displayed. However, I can double-click the file on the desktop and open it in LE. I can also save to the desktop even though the save panel does not show any files there.

However, if I close LE and then use Terminal, cd to the Desktop, and `open myfile.circ` it will open the file and now I can see the files using the open file dialog from within LE. In this case LE seems to be inheriting access permissions from Terminal.

I don't know if this oddity remains in later versions of macOS. But I'm pretty sure the problem is with access permissions rather than a problem in LE code.

davidhutchens, 2023-02-18:
After that comment, I launched LE from the Dock and tried to use the open panel again. This time it worked! I was playing with the permissions in System Preferences, Security & Privacy, Files and Folders. Now it behaves as expected showing the contents of the Desktop when permission is granted and not showing it when it is unchecked.

Whether checked or not, I can launch LE by double-clicking on the myfile.circ document.

I'm not sure what triggered the change in behavior. I had turned permissions on and off a few times before posting my previous note.

maehne, 2023-03-07:
Closing this as resolved due to no activity after @davidhutchens reply.

closed 2023-03-07 by maehne