logisim-evolution/logisim-evolution #1675: Documentation hardly/not readable with dark themes

R3dst0ne, 2023-02-21:
I noticed that the documentation is hardly readable, with some darker themes.

Here is how it should look like (`FlatLaf IntelliJ`):
[image]

`FlatLaf Dark` is hard to read:
[image]
And `GTK+` is impossible to read.
[image]

PhilipWaldman, 2024-03-21:
Is this issue still open? And if it is still open can I be assigned to it?

maehne, 2024-08-15:
Closing as PR #1986 got merged. Feel free to reopen if issue persists in other places.

closed 2024-08-15 by maehne