logisim-evolution/logisim-evolution #1682: Adding TTL 7487

aplq, 2023-03-05:
Short description: 4-bit True/complement, zero/one 
This would be a good 74 series chip for use behind an adder to set up for add/subtract and +1/-1 operations
Link to datasheet: [https://archive.org/details/bitsavers_tidataBookVol2_45945352/page/n403/mode/2up](https://archive.org/details/bitsavers_tidataBookVol2_45945352/page/n403/mode/2up)

I currently intend to add this myself, as it seems like an easy first part.

maehne, 2023-03-06:
@BunchOfBits has contributed over the past months a lot of TTL components to Logisim-evolution. You may review his [merged PRs](https://github.com/logisim-evolution/logisim-evolution/pulls?q=is%3Apr+is%3Aclosed) to get an idea what is need. He may be able to answer if questions should arise.

BunchOfBits, 2023-03-07:
Do you want me to add this one ? It shouldn't be much work, it is a very simple circuit

aplq, 2023-03-07:
Just created a pull request: Adding TTL 7487 #1684

BunchOfBits, 2023-03-07:
My question just got answered ;-)

PR #1684 (2023-09-09): Adding TTL 7487
This pull request is to close #1682
closed 2023-09-09 by BFH-ktt1