logisim-evolution/logisim-evolution #1685: Missing DLL

MinecraftOS, 2023-03-10:
I installed Java 19 and tried the .jar, which did not work. Then, I tried the .msi and that did not work either. It said there was a missing DLL.

R3dst0ne, 2023-03-11:
Could you please provide a view more information?
- What version of Windows do you use? (please include build number)
- What exact Java version do you use? (`java --version`)
- How did you invoke the `.jar`?
- Can you post the exact error message the `.msi` installer returns?

It would be great if you could try to build the 3.8.0 release and the current head of the develop branch.

MinecraftOS, 2023-03-13:
1. Windows 11 22H2
2. 19.0.2 2023-01-17
3. not sure what this means
4. attached image
[image]

maehne, 2023-03-13:
@MinecraftOS: What processor platform are you using to run Windows 11? x86_64 or arm64? Did you download the right MSI matching your processor platform? I.e., [`logisim-evolution-3.8.0-x86.msi`](https://github.com/logisim-evolution/logisim-evolution/releases/download/v3.8.0/logisim-evolution-3.8.0-x86.msi) for x86_64 or  [`logisim-evolution-3.8.0-arm64.msi`](https://github.com/logisim-evolution/logisim-evolution/releases/download/v3.8.0/logisim-evolution-3.8.0-arm64.msi) for arm64? Thes same holds true for the JDK. Is it matching your platform?

MinecraftOS, 2023-03-13:
Yes, the jdk is windows.
i used arm64.
i have a 12th gen intel core i5

MinecraftOS, 2023-03-13:
Oh boy. I am the stupid.

i used arm64. it should have been x86. pardon me.

maehne, 2023-03-13:
@MinecraftOS: No problem! I am glad that we could help you.

closed 2023-03-13 by maehne