logisim-evolution/logisim-evolution #1689: N type transistor outputs U or 0 for everything

MinecraftOS, 2023-03-17:
According to 
https://pages.cs.wisc.edu/~sohi/cs252/Fall2012/lectures/lec03_digital_logic.pdf
and
http://www.cburch.com/logisim/docs/2.7/en/html/libs/wiring/transist.html

the n type transistor _should_ act like a controlled buffer. However while designing my ALU i noticed that transistors do not work. Yes it is facing in the right direction.

Data | Gate | Q
0        0         U
1        0         U
0        1         0
1        1         U

Linux
AMD A8
openjdk 19.0.2

MinecraftOS, 2023-03-17:
another example
[image]

davidhutchens, 2023-03-17:
Please read the first document you posted from wisc.edu. You'll notice that for the N-type transistor, on page 3-5, it clearly says in the lower left of that page for the N-type transistor: "Terminal #2 must be connected to GND (0V)". Your examples have it connected to power. Note they use a triangle for ground and a T for power in their diagram while we use the triangle for power and a lined triangle for ground.

If you want a controlled buffer (or controlled inverted buffer) in your circuit, we strongly encourage you to use the provided gates rather than  build them with transistors.

MinecraftOS, 2023-03-17:
Hi david, indeed the paper states you need a ground I found this working as well as using a pull resistor.

R3dst0ne, 2023-03-26:
@MinecraftOS
If your Question got answered, please close this issue.
Otherwise, comment how we could help you.

MinecraftOS, 2023-03-29:
Its not answered: I have shown the N type transistor does not work, and I'm asking if there is anything you guys can do

R3dst0ne, 2023-03-29:
The transistors work as intended.
This is a common "problem" therefore it is mentioned in the [FAQ](https://github.com/logisim-evolution/logisim-evolution/wiki/Logisim-evolution-FAQ#transistor-behavior).
The Library Reference in LSe describes the behavior too. (The summarized form is not correct.)

@davidhutchens describes the behavior good in the issue #1295.
Other similar issues:
https://github.com/logisim-evolution/logisim-evolution/issues/1674
https://github.com/logisim-evolution/logisim-evolution/issues/1670
https://github.com/logisim-evolution/logisim-evolution/issues/1386
https://github.com/logisim-evolution/logisim-evolution/issues/1295
https://github.com/logisim-evolution/logisim-evolution/issues/504
https://github.com/logisim-evolution/logisim-evolution/issues/431

davidhutchens, 2023-03-29:
@MinecraftOS: Your posted screencast does not show the transistor to be broken. You have the output tied to ground, it should be the non-control input (lower one in your diagram) that is tied to ground, either directly, or though other n-type transistors. In cmos digital logic the n-type transistor is used to pull the output to ground or let it float. It is not used to push the output to one.

maehne, 2023-05-08:
Closing this issue as resolved due to inactivity of the original author. @MinecraftOS: Feel free to reopen if you have additional arguments.

closed 2023-05-08 by maehne