logisim-evolution/logisim-evolution #1712: Add the entries FpgaFreqPreDivider and FpgaFreqPreMultiplier as "# ==> FpgaFreqPreDivider =" and "# ==> FpgaFreqPreMultiplier =" in all the other fpga_*.properties files.

maehne, 2023-04-05:
> @chuckb : Thanks a lot for your contribution, I just forgot to ask you to add the entries FpgaFreqPreDivider and FpgaFreqPreMultiplier as "# ==> FpgaFreqPreDivider =" and "# ==> FpgaFreqPreMultiplier =" in all the other fpga_*.properties files. Could you please make a separate PR to fix this? Thanks!

Well, here is a question. I do not view them as FPGA properties, since that are not fixed per circuit. The base clock on the board is a fixed value; the technology is a fixed value. But the multiplier/divider are factors more akin to circuit parameters, but only apply during synthesis. So, where should I store them, if at all?

_Originally posted by @chuckb in https://github.com/logisim-evolution/logisim-evolution/issues/1702#issuecomment-1497505412_

chuckb, 2023-04-06:
I am happy to do this...I just want to make sure it's done in the right spot. I have not looked at where you are referring to, but I think these should be stored at the project level. Do you concur?

chuckb, 2023-04-10:
> I am happy to do this...I just want to make sure it's done in the right spot. I have not looked at where you are referring to, but I think these should be stored at the project level. Do you concur?

Oh, you may be referring to the localization on the GUI itself? I already did that (since I had already done it, it did not occur to me this is what you meant). I added the English and German strings. Is this what you meant? I have not added other languages.