logisim-evolution/logisim-evolution #1714: Nightly build is not triggered Nightly

R3dst0ne, 2023-04-06:
While looking at the [nightly builds](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml), I noticed that there is a build for each commit.
What is the status on this one?
Is this still wanted behavior?
https://github.com/logisim-evolution/logisim-evolution/blob/a8e3c096bbdad57975bbe4e8731c1e6cf5d09023/.github/workflows/nightly.yml#L13-L26

davidhutchens, 2023-04-06:
The problem, as I understand it, is that scheduled workflows only work on the default branch. We cannot easily schedule a nightly build that runs each night on the develop branch. Therefore, this issue is related to #1552. See my suggestion there.

davidhutchens, 2023-07-27:
Since we now have main as the default branch Nightly is running at midnight. PR #1796 adjusts it to only run at midnight and not run after each closed PR.

closed 2023-07-27 by BFH-ktt1