logisim-evolution/logisim-evolution #1716: Better Error Message when creating a new circuit

cybercoder-naj, 2023-04-08:
Instead of showing the required regex pattern, a message detailing why the name cannot be chosen is, in my opinion, a better user-experience. For instance, a message such as "Cannot start with a number" is a helpful message for those who not well-equipped with regex patterns.

[image]

maehne, 2023-06-07:
I agree, the actual rule is that it must be a legal identifier in VHDL and Verilog so that the name can be used without modification during the netlisting. A PR suggesting an improved error message, ideally including localisation strings is welcome.

davidhutchens, 2025-09-07:
I am closing this issue now. The PR claiming to fix it was merged 2 years ago and there has been no further discussion. If the problem remains, please reopen and describe the remaining problem.

closed 2025-09-07 by davidhutchens