logisim-evolution/logisim-evolution #1719: VHDL simulation doesn't work in Linux

gguarneri, 2023-04-10:
I'm trying to simulate a VHDL entity with Questa Advanded Simulator option setted in Logisim-evolution on Linux (Linux Mint). However, I received an error message in "VHDL simulator log" (see picture).

[image]

The Questa license file is good and correctly installed (see picture).
In Windows, this simulation is Ok.

gguarneri, 2023-04-11:
I found a workaround for this issue. I've installed Logisim-evolution from **Flathub**. This package distribution has the problem reported. I've uninstalled Logisim-evolution from **FlatHub** and installed it from the **DEB** package. Now it's working fine.

closed 2023-04-11 by gguarneri

maehne, 2023-04-11:
Yes, the sandboxing of Flatpak and Snap introduces limitations, which cannot be easily resolved. Therefore, we recommend using the native packages matching your Linux distribution. However, we are open to any PRs helping to resolve such issues.