logisim-evolution/logisim-evolution #1722: RAM completely broken in Logisim-evolution version 3.8.0

flzpfnove, 2023-04-10:
The output of the RAM is always colored red (color for when 2 components are writing to the same bus), even when no components other than the RAM are connected to the bus.

Image of the issue:
[image]

davidhutchens, 2023-04-11:
The RAM component is complicated because of its various attributes. You should check the Trigger attribute and the Asynchronous read attribute to make sure they are set as you desire. The default setting is non-asynchronous read which means a trigger, such as the default Rising Edge of the clock input, is necessary before an output is produced. Until then, the output is Error (E).

The connected RAM in your diagram is likely providing an Error output because it is waiting for the rising clock.

Note that your stand alone RAM must output Error since it has no address input.

If these comments do not help you, please provide more detail about the issue. Note you can post a Logisim-evolution .circ file by zipping it and then dropping it here if that will make your issue more clear.

flzpfnove, 2023-04-11:
I tried removing the controlled buffer from the RAM address bus so that it always gets an address, i tried modifying various attributes, and i also enabled the clock so it would get the rising-edge signals.

Image of the new slightly modified circuit and RAM component attributes:
[image]

Logisim circuit file:
[logisim_file.zip](https://github.com/logisim-evolution/logisim-evolution/files/11198514/logisim_file.zip)

closed 2023-04-11 by flzpfnove

flzpfnove, 2023-04-11:
Appearently enabling async read didn't actually fix the issue,

Zainabsoomro304, 2023-12-11:
@flzpfnove @davidhutchens could you figure out the issue?

davidhutchens, 2023-12-13:
@Zainabsoomro304:  Issue #1747 is related to this problem. PR #1824 presumably fixed it. Could you try the [nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml) to see if it resolves your issues. If not, open another issue describing what is happening and what should happen. Note that you may need to look down the list of nightly builds to find the last one that actually builds the packages since it skips the build if nothing has changed.

Zainabsoomro304, 2023-12-14:
@davidhutchens I'm just a beginner, couldn't understand the solution. could you elaborate on it?

davidhutchens, 2023-12-14:
The nightly build is a build of our current work. Follow the link I gave you. Look down the list you will find there to find the latest build that contains a package for your system. Download and Install that package and see if it solves your problem.

If not, create a new issue and describe your problem as clearly as you can. You can include a .circ file if you zip it first.

mags0ft, 2024-04-14:
I've tried to use the RAM too and can confirm that it is even more broken than that, as it disappears completely when you try to move it or use CTRL+Z. It won't come back, even with CTRL+SHIFT+Z. As if it wasn't there, like a ghost component...