logisim-evolution/logisim-evolution #1724: Feature Request: Export image: add PDF format

45gfg9, 2023-04-11:
[image]

Add an option to export as PDF here

BFH-ktt1, 2023-04-17:
Dear @45gfg9 : Thank you very much for your request. Unfortunately we are quite limited in resources for the moment, such that I do not think that we will tackle this feature in the near future. Any PR's in this direction are of course welcome.

45gfg9, 2024-08-27:
Looking back in 16 months, I think this can be closed since it already supports Ti*k*Z output, and can be used to generate PDF / EPS format. Some tweaks to the generated code are needed (colors, fonts, etc.) but it will do.

closed 2024-08-27 by 45gfg9