logisim-evolution/logisim-evolution #1725: PR #1704 breaks controlled buffer (and inverter) if it is set to more than 1 bit

davidhutchens, 2023-04-11:
PR #1704 uses code that contains `in == Value.TRUE || in == Value.FALSE`, `in == Value.UNKNOWN`, and `in == Value.ERROR`. That works great if `in` is a single bit. But it is meaningless when `in` is wider than 1 bit.

I think I see how to fix it.

Assuming our hardware folks are happy with the U being passed through the controlled buffer, should it also be passed through the controlled inverter? Our Buffer passes it through, but our NOT gate does not and produces E on U input. I'm not a hardware person, so I am not sure what the desired result actually is. The earlier PR assumes the inverter also passes the U through.

@maehne and @BFH-ktt1, please advise.

prlaba, 2023-04-14:
Why shouldn't the behavior of the Controlled Buffer component, when enabled, be any different than the Buffer component's behavior?

The Buffer component passes its input (0, 1, U or E) to its output, unchanged.  It seems inconsistent to me to have the Controlled Buffer component, when enabled, behave any differently.

I agree that the Controlled Buffer/Inverter component, when enabled, should behave the same as a NOT gate.

davidhutchens, 2023-04-14:
@prlaba: I take your comment to be a vote for choice 3) in my list. The more I think about it, the more I lean that way myself. I would still like to hear from @maehne and/or @BFH-ktt1.

prlaba, 2023-04-14:
Yes, I agree; your choice 3 is the best option.

This will break backward compatibility, although I don't expect there are many (any?) circuits that are dependent on an Error output from the Controlled Buffer component.

Thanks,

seanpquinn, 2023-04-15:
If choice 3 is agreed on, have a potential patch for this in #1726 if you'd like to take a look

Sorry for the inconvenience again on the breakage

BFH-ktt1, 2023-04-17:
As #1726 has been merged I close this issue, please reopen if required

closed 2023-04-17 by BFH-ktt1