logisim-evolution/logisim-evolution #1733: Select/Enable pin placement bug with "narrow" Multiplexer

xarkenz, 2023-04-25:
This combination is very specific, but in my experience, very easy to run into. When a Multiplexer has "narrow" size, top/right select location, and 1 select bit, the Select pin (plus the Enable pin if, well, enabled) appears 1 grid unit higher than it should, and is then visually detached from the component.
[image]
I suspect the reason for this bug is that the "narrow" Multiplexer's select/enable pins are always 1 unit above where the "wide" pins would be relative to the anchor, and that the code doesn't account for the fact that the 1-bit "wide" multiplexer appears much different from the 2-bit, etc. versions (as seen in the image). I plan on investigating this to see if I can find the problem and fix it.

Although, one thing that worries me is that this is actually an issue with the pin placement, and as such a fix would break existing circuits using a Multiplexer exactly like this...

maehne, 2023-04-25:
I agree, this inconsistency should be fixed.

maehne, 2023-04-29:
Closing as fixed by merged PR #1734.

closed 2023-04-29 by maehne