logisim-evolution/logisim-evolution #1738: random delay option results in broken negating-gate that outputs 0 for 0 on input

horseprint, 2023-04-28:
The bug is that

1) sometimes negate gate (NOT gate) fails in random delay mode. Note that the input is not an Error nor an Unknown, it is just 0, and output is also 0... I do not see how such state should ever arise even in random delay mode (random sequence of propagation of signals in a clock cycle?).

2) either that is a bug in simulation - or otherwise the effects of "random delay" (in project settings) are no where explained what they actually do and what to expect, and do they uncover problems in design.

Example circuit attached.
Is it designed properly for the function it has (it should count startup in init steps called by me "runlevel" like 0,1,2 and then should increase register IP 0,1,2,3,4 and on IP==4 it simulates instruction to reset this mini-computer that is re-runs the init steps which include setting back IP to value 0)?
Is the "random delay" function uncovering an actual flaw in design of this circuit (that can be fixed - how?) or is it leading to a bug in logisim?

Either way it feels like the "random delay" function needs a bit more documentation.

### Description of attached circuit

I have a problem with logisim and a circuit: when running in random-delay of gates mode, sometimes the negating-gate is giving same output as input instead of negating it, as demonstrated by attached circuit, and the image screenshot with parts marked in colors.

This circuit starts up by counting the startup/reset steps in runlevel_ix register from 0 to 3, and then this microcomputer works by increasing the IP counter, up until IP==4 then it simulates instruction to reset itself by setting rstC=1.

Logic of “resetting” works like this:

    code decides it is time to reset, CPU (BLUE) sets cmdC=1. In this example we do not have ROM, just a simple comparator so that when IP==4 then we set cmdC=1 which is what reset opcode would do.

    this results in GREEN circuit setting reg_run=0 (“is running of CPU allowed”) which forces reset of register runlevel_ix=0.

    runlevel_ix==0 means machine start rebooting, it increments itself, at runlevel_ix==1 it sets rst=1, which results in IP register being reset to value 0.

The problem is: after some time, the gate “negate” fails. In the screenshot, the first resetting impulse step is complete, and we should allow runlevel_ix to incremenet up, but since the “negate” node failed the neg_after==1 so then the runlevel_ix is not allowed to increment and freezes at current value.

[image]

The text file should be renamed to ".circ" 
[timers-bug2mini04.txt](https://github.com/logisim-evolution/logisim-evolution/files/11355703/timers-bug2mini04.txt)

horseprint, 2023-04-28:
in Logisim-Evolution 3.8.0

maehne, 2023-04-30:
When I look at your screenshot, I notice that you use a gated reset signal to *asynchronously* reset your counter  on the right. This typically leads to unwanted/surprising behaviour. I suggest that you first fix this issue by using a synchronous clear (add a multiplexer in front of the registers D input that either forwards the incremented value or the reset value) and then check whether this problem persists.

maehne, 2023-05-08:
I can reproduce the issue with the occurence of inconsistent input and output values on the inverter chain. However, I am unsure this can be solved reliably in Logisim-evolution's simulation kernel without major changes/rewrite. Logisim-evolution's simulation algorithm is not perfect, as documented in the Help -> User Guide -> Value Propagation. Its grasp of time and gate delays is limited and works well for synchronous circuits, but not digital circuits, which exploit asynchronous effects and gate delays. In these cases, you are better off simulating your circuit with a mature HDL discrete event simulator or even SPICE (without any guarantee that your manufactured circuit will work reliably).

maehne, 2026-09-28:
I agree to close this issue as resolved by the documentation update. The delay component (or rather extension of the Buffer component with a configurable delay) should address @horseprint's use case if we should decide to implement it.

@horseprint: Feel free to comment to this issue and issue #1840 to clarify further your needs.

closed 2026-09-28 by maehne