logisim-evolution/logisim-evolution #1739: Add language support for Hebrew

YovelB, 2023-04-29:
A few colleges and universities use the tool in Israel, mainly in Electronics Engineering.

I would to add support for the Hebrew language 
Do I have permission to do so?

maehne, 2023-04-29:
Of course! Such contributions are welcome! Just create a PR against the branch `develop`. If you want early feedback on your work, you can mark the PR as a draft and already push an incomplete state, which you then complete with additional commits.

maehne, 2023-04-29:
One challenge, which might require some investigation from your side is how to handle the different writing direction.

YovelB, 2025-07-26:
Thanks for the feedback @crossplatformdev. When I tried dealing with this issue I had problems understanding swing and the project base code. So, I have abandoned it, without making it clear (my fault).

If someone volunteers to deal with this issue it would be great. I would be happy to help in the locale and formatting side. Else, feel free to close the issue.

crossplatformdev, 2025-07-30:
Hi @maehne ,

What about simply writing the strings reversed in the properties Hebrew locale strings file?

Will be very simple workaround, just in case the user needs it.

Kind regards,
Elías A.

YovelB, 2025-11-13:
I dont see this as important enough, closing issue.