logisim-evolution/logisim-evolution #1743: Color Picker is not translatable

smallg0at, 2023-05-02:
Under the `src\main\resources\resources\bric` directory are multiple color picker localization files, but they seem to be unused.

It would be favorable for localization tasks if the additional `ColorPicker_**.properties` files actually functions, or put those strings into the localization directory for translation.

maehne, 2023-05-08:
While reviewing PR #1740, I came also across this issue and looked a bit into it. I think that `src/main/resources/resources/bric/` for `ColorPicker` is a leftover, which can be removed, because `build.gradle.kts` adds this module as an external dependency, which gets downloaded by Gradle during build. So, the localization strings would need to be submitted upstream to have any effect.

@davidhutchens: Do you agree with my assessment? Could you maybe also have a look at `src/main/resources/resources/connectina/`. The therein contained `FontChooserDialog` / `JFontChooser` seems not to be referenced elsewhere. Am I right or am I overlooking something?

davidhutchens, 2023-05-09:
@maehne: I agree with your assessment about ColorPicker. The `org.drjekyll:colorpicker` dependency has its own language resources. I tried removing the `src/main/resources/resources/bric/` directory and compiling. The ColorPicker works in languages they have included before and after removing the bric directory. I think the bric files can be removed from our source. To add a language, a PR would need to be directed upstream.

I also looked for any use of the connectina FontChooser resources and I didn't find any. Our FontSelector works the same in various languages before and after the connectina directory is removed. I see no reason we need it.

I did notice that the FontSelector's `Bold` and `Italic` checkbox labels seem to remain in English, no matter what language is selected.

closed 2023-08-29 by BFH-ktt1